espressif/idf-eclipse-plugin #275: How to change the flasher speed

icodk, 2021-05-25:
The plugin (2.1.0.202104161244) is flashing new program using -b 460800
How can I change it to -b 921600 ?

Maybe to add it as a parameter to the serial flasher configuration ?

kolipakakondal, 2021-05-25:
Hi @icodk 

This can help you https://github.com/espressif/idf-eclipse-plugin#launch-configuration

In the arguments section, you need to pass the complete command with the baud rate.

icodk, 2021-05-26:
I tried with 
Location:
```
${system_path:python}
```
Working directory:
```
${workspace_loc:/esp32_lvgl}
```
Arguments:
`esptool.py --chip esp32 -p COM5 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 lvgl-demo.bin`

I got the following error:
`C:\Dev\esp32-idf\idf-tools\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe: can't open file 'esptool.py': [Errno 2] No such file or directory`
What am I doing wrong ?

kolipakakondal, 2021-05-26:
`esptool.py` - this requires the absolute path, could you check that?

I know this restriction from a plugin, will enhance this.

icodk, 2021-05-26:
Would love to give absolute path for esptools.py. But have no idea where esptools.py is located.
Is it part of the esp-idf ?

icodk, 2021-05-26:
YES !!! 
This did the trick:
In the Arguments field I entered :
`C:\Dev\esp32-idf\esp-idf-v4.2\tools\idf.py  -p COM5 -b 921600 flash`

Thanks

closed 2021-05-26 by icodk

f-hoepfinger-hr-agrartechnik, 2021-06-30:
i do it that way: 
esp-idf-v4.3\tools\idf_py_actions\serial_ext.py

change Line 146.

in my Opinion easier.