espressif/idf-eclipse-plugin #279: My purely professional assessment: the plugin is not operational

mickpf, 2021-05-31:
I am frustrated!

I don't understand why you make software (both the plug-in for Microsoft Visual Studio Code and the plug-in for Eclipse for Ubuntu 64 bit) available on the Internet and explain it to be operational, even though they are actually still completely immature.

I don't really want to talk about the plug-in for vscode at this point, because it works at least for the most part. The communication with the ESP32 and the debugging (with openocd) via the JTAG adapter do not work at all. I can flash ESP32 via the JTAG adapter (through both serial ports) without any problems (by the 'idf.py flash' command) and communicate with the ESP32 (by the 'idf.py monitor' command), but not from within vscode!

But now to the plug-in for Eclipse:

I have already tried several times following the instructions to install Eclipse-CDT (2020-12 and 2021-03 with updating eclipse before starting plug-in installation) and the plug-in for Eclipse, but in the end the plug-in does not work at all!

ESP-IDF and the tools must be reloaded from the Internet during the installation process, even though they are already installed in the correct version on the computer. You can specify that the already installed ESP-IDF should be used, but that doesn't work at all afterwards. The installation program does not give the possibility to use existing tools (in the "$ IDF_TOOLS_PATH"), it MUST download them from the Internet, and installs them anywhere in a (unknown) path where you do not want to see/have them! It should be possible to use the tools you already have installed!

And the whole resulting installation don't work at all! I can create a ESP-IDF project, but it is not possible to "Launch Target" because this toolbar does not apear on the screen and it is not possible to activate it. Without this step it is not posiible to compile any project!

I'd like to debug code in a modern way with a GUI, gdb adn/or openocd and without using ESP_LOGI, but it seems to be imposssible!

Kind Regards,
Michael

kolipakakondal, 2021-06-01:
Hi @mickpf 

Thanks for the report and sorry for the trouble you had to go through. Please provide the plugin version and log to troubleshoot this problem.

On the VSCode debugging via JTAG issue - I have spoken to my colleague, he will take a look at it.

On the Eclipse plugin issue - Let us verify the latest plugin with Eclipse 2021-03 and see if we reproduce the issue on Ubuntu 64 bit.

kolipakakondal, 2021-06-01:
> ESP-IDF and the tools must be reloaded from the Internet during the installation process, even though they are already installed in the correct version on the computer. You can specify that the [...]

If you have followed the "Install tools" process from Eclipse, this will take care of getting the right set of tools based on the selected esp-idf version. Could you please provide the installation log and error which you are facing?

> And the whole resulting installation don't work at all! I can create a ESP-IDF project, but it is not possible to "Launch Target" because this toolbar does not apear on the screen and it is not possible to activate it. Without this step it is not posiible to compile any project!

Launch target toolbar does not appear issue comes when Eclipse CDT is installed partially or installation is broken. Could you take the latest Eclipse CDT package from here and see https://www.eclipse.org/downloads/packages/release/2021-03/r/eclipse-ide-cc-developers

Let me see if I can provide the installation screenshot from the Ubuntu envionment.

brianignacio5, 2021-06-01:
With regards to vscode extension, you say that is doesn't work but could you provide any error or feedback to understand the issue you are having ? 

Did you read the extension [documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) or [install tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) to configure the vscode extension?

mickpf, 2021-06-01:
Hello Kondal,
many thanks for your answer! I didn't hope that anyone would respond to my report!
The Eclipse plugin version: 2.1.0.202104161244.

I had allowed the installation process to install the tools even though a previous installation attempt had already placed the correct version of the tools in that path.
Every time at the end of the installation appeared the list of the installed tools with status "installed".
The "ESP-IDF Tools Manager -> List installed tools" displays the same list.

When I install development tools such as Eclipse or ESP-IDF, I always take great care to ensure that the installation process runs without errors and that the correct version is downloaded and installed. My Ubuntu is also up to date.
The only thing I noticed: when I installed the plugin, a list of two additional packages appeared in the window. One of them  labeled something with "... Codered ..." had a red tick on the left. That means it cannot be installed.

Kind Regards,
Michael

mickpf, 2021-06-01:
Hello Brian,

First of all, I would like to apologize and/or correct myself.

After numerous installation attempts, suddenly since yesterday the communication through the JTAG adapter works. I don't know why, but it works now. I can now flash and monitor the device(s).
Following the insttructions I have to install the "esp-debug-adapter" now, I've downloaded it (through "git clone ..."), but I cannot find any installation instructions for it!

Kind Regards,
Michael

sigmaaa, 2021-06-01:
Hi @mickpf,

can you provide the product information log from the eclipse, please? This will help us to reproduce the problem. 
To do in , in eclipse, go to Help > Product information.

The next thing you can try to do to make the launch target toolbar appear:
go to Help > Install New Software... > in the field "Work with" select CDT (for example CDT - http://download.eclipse.org/tools/cdt/releases/10.1) and then select and install LaunchBar

[image]

Best regards,
Denys

mickpf, 2021-06-01:
Hello,
It is difficult, both for me and for you, to deal with two problems in one ticket.
Could you please set up a second ticket for VS Code and only cover the problems with the Eclipse plugin here?

The "Product Information" displays:
`Executing python /opt/ESP32/v4.2.1/esp-idf-v4.2.1/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

IDF_PATH: /opt/ESP32/v4.2.1/esp-idf-v4.2.1
IDF_PYTHON_ENV_PATH: <NOT FOUND>
PATH: <NOT FOUND>

Operating System: linux
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF version cannot be checked. IDF_PATH or IDF_PYTHON_ENV_PATH  are not set.
Python set for IDF_PYTHON_ENV: <NOT FOUND>
Python declared in PATH variable: Python 3.8.5
`
I've tried to set "IDF_PYTHON_ENV_PATH" to "/opt/ESP32/v4.2.1/python_env/idf4.2_py3.8_env" (with rebooting the computer), but it does not work! How should I set this?

I followed your hint with the launch bar. There are only two modules available: "LaunchBar Developer Resources" and "Remote Launch Targets Developer Resources". I'll install them and give you a response...

BTW. Regarding the VS Code plugin: the "esp-debug-adapter" was already installed with the plugin. I don't need to install it. I'll go through the instructions further...

Kind Regards,
Michael

mickpf, 2021-06-01:
Hello,

I installed the Launch Bar and restarted Eclipse, but as soon as I activate it under Preferences, Eclipse becomes extremely slow. Restarting Eclipse does not bring any improvement.
I'm wasting too much time trying to get this plugin running.
Since it does not seem possible to get the Eclipse plugin to work with acceptable effort, I would like to stop at this point and concentrate on the VS Code. Except for debugging, it works.

Still, thank you very much for your help.

Kind Regards,
Michael

closed 2021-06-01 by mickpf