espressif/idf-eclipse-plugin #282: New Eclipse plugin shows randomly unresolved types, vars members etc.

samsam4, 2021-06-02:
**Describe the bug**
New Eclipse plugin shows randomly unresolved types, vars members etc.in Problems View while build output in Console shows always: Build complete (0 errors, 0 warnings)

And the Problems alternate with different amounts between 1 and 16 on every next press of the Build button.

In the original code that I downloaded from [https://github.com/nopnop2002/esp-idf-ili9340](https://github.com/nopnop2002/esp-idf-ili9340)
while trying to solve the problem, I added   #include "dirent.h"   but the output is absolutely same as output without rhis extra line . The compiler/indexer sure sees the dirent.h, because I tried with slightly modifying the name and get error mark in front of it right away. 
Still cant find why cant see the DIR type defined there (always) and Fields under dirent* pe on every even try to Build (once build no problem with those fields, next build 15 more problems with those fields, next build again no problem with those fields (only DIR type problem - always)?  

The same code compiles, build and flash from CMD environment without any reported problems, this  only happens in Eclipse!

**To Reproduce**
Steps to reproduce the behavior:
1. Downloads project from here:
[https://github.com/nopnop2002/esp-idf-ili9340/archive/refs/heads/master.zip](https://github.com/nopnop2002/esp-idf-ili9340/archive/refs/heads/master.zip)
2. Import the project
3. Build
(4. )Set sdkconfig for TFT Configuration as per last screenshot
(5. )Build again
6. Observe Console and Problems windows
7. Build again 
8. Observe Console and Problems windows

**Expected behavior**
No errors reported in Problems window or at least same number errors reported on every next Build without modifying anything in code and/or settings.  

**Screenshots**
[image]

[image]

[image]

**Environment Information:** 

Operating System: windows 10

Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
(installed with 2.9beta online installer)

**Eclipse Error log:** 
[errLog1-6-21_2.log](https://github.com/espressif/idf-eclipse-plugin/files/6581327/errLog1-6-21_2.log)

Last operations in the attached error log errLog1-6-21_2.log were:
1. Clean the project
2. Tried to delete (unsuccessfully) the 'build' folder from Eclipse
3. Closed Eclipse
4. Deleted 'build' folder with Windows Explorer
5. Start Eclipse
6. Build the project - receive only 1 problem - unresolved type DIR
7. Build again - received 16 errors (DIR + 15 Fields)

Thanks

kantrol, 2021-06-17:
Same issue here on linux

Operating System: linux
Java Runtime Version: 15.0.3+3-Debian-1
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF v4.4.0
Python set for IDF_PYTHON_ENV: Python 3.9.2

pirobk, 2021-11-26:
A bit more info on this.

The main issue seems to be the evaluation order of system includes, which have multiple variants.

E.g. the sys/dirent.h is empty in xtensa-esp32-elf/sys-include/sys/dirent.h but the variant used during build by the compiler is from the newlib component's include folder components/newlib/platform_include/ which correctly defines the missing types.

kolipakakondal, 2022-02-14:
Hello All, Indexer and unresolved header issues are fixed as part of the IEP 2.4.0 release. Please get the latest changes using the update site to verify. Thank you.

closed 2022-02-14 by kolipakakondal