espressif/idf-eclipse-plugin #283: Bug: Eclipse plugin occasionally missing to interpret PATH variable and cant find GIT and Python

samsam4, 2021-06-02:
**Describe the bug**
Initially after installation online installer 2.9beta report for Help>Product Information was Ok (see [https://github.com/espressif/idf-eclipse-plugin/files/6535933/err_log_blink2.log](https://github.com/espressif/idf-eclipse-plugin/files/6535933/err_log_blink2.log)), couple days after (without any changes in installation or settings) start reporting:
...
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
... 
Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

Same problem was observed and before with 2.6 offline installer (see [https://github.com/espressif/idf-eclipse-plugin/issues/259#issuecomment-831371038](https://github.com/espressif/idf-eclipse-plugin/issues/259#issuecomment-831371038)  )

2-6-21

Help>Product Information:

Executing D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe D:\ESP2_9b\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: D:/ESP2_9b/
IDF_PYTHON_ENV_PATH: D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env
PATH: D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;D:\ESP2_9b_tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\cmake\3.16.4\bin;D:\ESP2_9b_tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;D:\ESP2_9b_tools\.espressif\tools\ninja\1.10.0\;D:\ESP2_9b_tools\.espressif\tools\idf-exe\1.0.1\;D:\ESP2_9b_tools\.espressif\tools\ccache\3.7\;D:\ESP2_9b_tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;**D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts**;D:\ESP2_9b\tools;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin/server;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin;D:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\;D:\ESP2_9b_tools\.espressif\tools\idf-git\2.30.1\cmd\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\ActiveState Komodo Edit 11\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;D:\AndroidSDK\platform-tools\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Users\xx\AppData\Local\Microsoft\WindowsApps;;D:\ESP2_9b;

Operating System: windows 10
Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
ESP-IDF v4.2.1
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

**The only work around found so far:**

Add to Path Windows Environment Variable (exact same paths copied from PATH: above):
D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts
D:\ESP2_9b_tools\.espressif\tools\idf-git\2.30.1\cmd\

Restart Eclipse:
Help>Product Information:

Executing D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe D:\ESP2_9b\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: D:/ESP2_9b/
IDF_PYTHON_ENV_PATH: D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env
PATH: D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;D:\ESP2_9b_tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\cmake\3.16.4\bin;D:\ESP2_9b_tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;D:\ESP2_9b_tools\.espressif\tools\ninja\1.10.0\;D:\ESP2_9b_tools\.espressif\tools\idf-exe\1.0.1\;D:\ESP2_9b_tools\.espressif\tools\ccache\3.7\;D:\ESP2_9b_tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;D:\ESP2_9b\tools;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin/server;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\ActiveState Komodo Edit 11\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;D:\AndroidSDK\platform-tools\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts;D:\ESP2_9b_tools\.espressif\tools\idf-git\2.30.1\cmd\;C:\Users\xx\AppData\Local\Microsoft\WindowsApps;D:\ESP2_9b;D:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\

Operating System: windows 10
Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF v4.2.1
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.8.7

**To Reproduce**
Steps to reproduce the behavior:
1. Help>Product Information
2. If reports problems with GIT and Python in PATH variable: add same paths in Windows System Environments variable Path 
3. Restart Eclipse
4. Help>Product Information

**Expected behavior**
Paths to git and python declared in Eclipse PATH variable should be enough to function properly without need to duplicate in Windows 'Path'  environment variable

**Screenshots**

[image]

**Environment Information:** 
Operating System: windows 10
Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
(used online installer 2.9 beta)

**Eclipse Error log:** 

[errLog2-6-21.log](https://github.com/espressif/idf-eclipse-plugin/files/6584904/errLog2-6-21.log)

samsam4, 2021-06-13:
After heavy fight with Visual Studio Extension it finally make my previous IDF/tools installation (2.9beta online installer) completely useless, so I had to make a new clean installation (this time with 2.8 offline installer so to have easier way to recover from damages and compare to find the root cause if possible). So I have some new data that might help you better source the problem:

When I first **install 2.8 (with ESP-IDF 4.2.1) and immediately start Eclipse** after - the report from Help>Product information was Ok.  
Even I compile  the 'blink' example without any problems.

[ProdInf2_8-13-6-21_1.txt](https://github.com/espressif/idf-eclipse-plugin/files/6644101/ProdInf2_8-13-6-21_1.txt)
[err_log13-6-21_10am.log](https://github.com/espressif/idf-eclipse-plugin/files/6644104/err_log13-6-21_10am.log)

Then I **restart computer and start Eclipse** and run Help>Product information and again as previously reported - git was not fond and python in the PATH was not found. 

[ProdInf2_8-13-6-21_2.txt](https://github.com/espressif/idf-eclipse-plugin/files/6644106/ProdInf2_8-13-6-21_2.txt)
[err_log13-6-21_10-20am.log](https://github.com/espressif/idf-eclipse-plugin/files/6644108/err_log13-6-21_10-20am.log)

Those are only 2 errors/differences in Product information report:

Python declared in PATH variable: **Python 3.8.7**

vs.  :

error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
...
Python declared in PATH variable: **Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases**.

Paths to python and git reported are absolutely same in both scenarios. Let me know if there is a way from eclipse to see in what environment: paths, env. variables is running Eclipse, Python or whatever is involved in processing Help>Product information, so I could also try to track route cause of the problem?

Hope that will help you fix this annoying  problem, let me know ASAP if you need any additional information from me, because when start playing with VSC extension most probably and this installation will stop working?

Thanks

samsam4, 2021-06-27:
I've made one more clean installation with Installer v2.8 (with ESP-IDF v4.2.1), so now can give more definitive report when the installation Eclipse/ESP-IDFtools works OK: only when Eclipse is started with marking 'Run ESP-IDF Eclipse Environment' at the final  screen of the Installer:

[Help_ProductInformation1.txt](https://github.com/espressif/idf-eclipse-plugin/files/6720899/Help_ProductInformation1.txt)
[err_log1.log](https://github.com/espressif/idf-eclipse-plugin/files/6720900/err_log1.log)

Once this first Eclipse run is closed and started again (from the shortcut icon) even in the same Windows session (no Windows restarts or any other program run between the two Eclipse sessions) and no more git and python are recognized/ found in the environment path:

[Help_ProductInformation2.txt](https://github.com/espressif/idf-eclipse-plugin/files/6720902/Help_ProductInformation2.txt)
[err_log2.log](https://github.com/espressif/idf-eclipse-plugin/files/6720903/err_log2.log)

Hope that would help you easier and quicker identify and fix the problem ;)
Regards

kolipakakondal, 2021-06-29:
Hi @samsam4 I am able to identify the problem, let me provide a patch version soon. However, this should not impact any other functionality of the plugin other than the `Help > Product Information`.

Thanks for reporting and providing the logs

samsam4, 2021-06-29:
@kolipakakondal  The problem that current version of Eclipse plugin cant find paths to git and python is not affecting only Help > Product Information:
(unless git and python paths are added to the Windows Path env variable, whole bunch of problems follow)

First  serious problem is that environment variables will not be filled properly:
[image]

Second problem of course is that without those variables (as could be expected) the project cant be compiled/build successfully: 'No Toolchain found for Target esp32' , etc.
The log below is from a new project from template  'blink':
[eclipse28-6-21.log](https://github.com/espressif/idf-eclipse-plugin/files/6730354/eclipse28-6-21.log)

kolipakakondal, 2021-06-29:
> @kolipakakondal The problem that current version of Eclipse plugin cant find paths to git and python is not affecting only Help > Product Information:
> (unless git and python paths are added to the [...]

> Second problem of course is that without those variables (as could be expected) the project cant be compiled/build successfully: 'No Toolchain found for Target esp32' , etc.
> The log below is from a [...]

This got fixed with the eclipse plugin 2.1.1 https://github.com/espressif/idf-eclipse-plugin/issues/259

samsam4, 2021-06-29:
Here is how blink example Environment and error log look like when Windows Path includes paths to git and python (build finishes: Build complete (0 errors, 0 warnings): D:\W21\ESP\wsp21-4\blink\build)

[image]
[eclipse28-6-21_2.log](https://github.com/espressif/idf-eclipse-plugin/files/6730642/eclipse28-6-21_2.log)

Quite surprising is how Eclipse behaves when it started like georgik proposed [here](https://github.com/espressif/idf-installer/issues/28#issuecomment-869687631)

In Environment there is only   IDF_PATH D:/ESP2_8/esp-idf/
but build passes and finishes with 0 errors, 0 warnings???

samsam4, 2021-06-29:
@kolipakakondal  'This got fixed with the eclipse plugin 2.1.1 #259'

Unfortunately I cant agree at all. I've made new clean installation with v2.8 offline installer, updated plugin to last available v2.1.2 and the mess with environment variables is same as [before (v2.6)](https://github.com/espressif/idf-eclipse-plugin/issues/259#issue-874593713) maybe even worse :(

This is how looks Environment with new Plugin v2.1.2:

[image]

Cant start a New ESP-IDF project as it asks me to set paths and Install (again tools):
[image]
[image]
[image]

Even if I set manually the paths (which should be done from the plugin as paths are anyway available in the D:\ESP2_8\.espressif\esp_idf.json file) and have output from Install tools with plenty of 
`FileNotFoundError: [WinError 2] The system cannot find the file specified`
[ConsoleOutput.txt](https://github.com/espressif/idf-eclipse-plugin/files/6733647/ConsoleOutput.txt)

At the end Environment looks absolutely the same as before and I cant start a new project, because I'm cycling same errors again and again :(
[err_log29-6-21_1.log](https://github.com/espressif/idf-eclipse-plugin/files/6733665/err_log29-6-21_1.log)

______________________

While if I add paths to git and python in Windows Path env variable, the situation is much  better:

[image]
but not enough though :( , because when try to start a new ESP-IDF project again starts to ask me to Install Tolls (despite both IDF_PATH and PATH are available):

[image]
[image]
git path again have to be filled manually :(

[image]
[ConsoleOutput14.txt](https://github.com/espressif/idf-eclipse-plugin/files/6733750/ConsoleOutput14.txt)

At least at the end after so unnecessary steps that suppose to be done automatically from the plugin I can start and build a new project from 'blink' template:
[err_log29-6-21_2.log](https://github.com/espressif/idf-eclipse-plugin/files/6733759/err_log29-6-21_2.log)

samsam4, 2021-06-29:
Starting Eclipse from powershell(ESP-IDF 4.2 PowerShell) with: `& '~\Desktop\ESP-IDF Eclipse.lnk'` without git and python paths existing in the Windows Path env variable, and trying to start a new ESP-IDF projects goes trough the same steps and screens (pictured in my previous post - first 4 pictures)

The only difference is the output from 'Install tools':
[ConsoleOutputPShEcl.txt](https://github.com/espressif/idf-eclipse-plugin/files/6734820/ConsoleOutputPShEcl.txt)

and at the end again cant start new  project and every next attempt push me to the same steps :( :(
[err_log29-6-21_3.log](https://github.com/espressif/idf-eclipse-plugin/files/6734823/err_log29-6-21_3.log)
[ProductInformationPShellEcl1.txt](https://github.com/espressif/idf-eclipse-plugin/files/6734826/ProductInformationPShellEcl1.txt)

sigmaaa, 2022-05-09:
closed as not reproducible on the latest versions of the plugin. Looks like it was fixed during some other ticket

closed 2022-05-09 by sigmaaa