espressif/idf-eclipse-plugin #284: Launch Configuration is saved in the Workspace and not in the project

icodk, 2021-06-03:
I recently cloned a project from github, and found out that my Launch configuration changes are left behind in  the
original location of the project and the original Workspace but not as part of the project that I pushed to github
This is probably an architectural decision, which in my opinion is wrong because Launch Configuration has nothing to do with  the workspace and everything to do with a project. 
The exact location is a file named: `project_name.launch` located in  :
`\.metadata\.plugins\org.eclipse.debug.core\.launches`
I hope there is a way to change it.
STM32CubeIDE is also an eclipse based IDE, where the launch configuration is saved as a file in the project root, makes it easier to keep the project self contained without bloating it with  workspace .metadata  which can be common for different unrelated projects

kolipakakondal, 2021-06-04:
Hi @icodk 

Thank you for your suggestion, will look into it.

gregtoth, 2022-10-17:
In the launch configuration, have you tried changing the Common > 'Save as' setting to 'Shared file' instead of 'Local file'?