espressif/idf-eclipse-plugin #295: Feature requests: line width and total number of lines

liudr, 2021-06-14:
1. Line width is fixed at 80. This is in contrast with most screens that can display hundreds of characters and clips the messages unnecessarily. Please add an option to change line width or fit line width within the display area.
2. Total number of lines. Currently it is fixed at 1000 lines. Most terminal programs offer an option to increase this. Suppose I want to log the printouts for some extended period of time or log lots of outputs that are printed quickly, 1000 lines is not enough. There is no real reason to cap the lines at 1000. Have a much larger number say 65536 or have an option to customize it.

fhgeoprospectors, 2021-06-30:
totally agree, that 80 char wide is old-tech. it should be unlimited OR adjustable.

kolipakakondal, 2021-07-01:
Hi @liudr @fhgeoprospectors Fix is available currently on the latest build of the master branch. This is not yet available in the stable release.

If you would like to try please pick this build https://github.com/espressif/idf-eclipse-plugin/suites/3120488433/artifacts/71485600

Install as per the instructions here https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

And configure as per this https://github.com/espressif/idf-eclipse-plugin#esp-idf-serial-monitor-settings

alirana01, 2021-08-16:
@liudr The feature is now part of the latest release
[image]
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0

closed 2021-08-16 by alirana01