espressif/idf-eclipse-plugin #298: IDF-tools installation is saved to workspace instead of the plugin configuration

icodk, 2021-06-19:
I recently cloned an idf-esp plug-in project on a computer that already have eclipse and idf-plugin installed.
Trying to build the project, I got the error: `No Toolchain found for Target esp32`
However, the tool chain was working fine in a nother workspace.
Looking at Help->ESP-IDF Tools manager->Install Tools,  There are 3 fields:
```
1. ESP-IDF Directory
2. Git Executable Location
3. Python Executable location
```
Field 1 was still pointing to the original ESP-IDF directory. The other two fields where empty.
Filling the two other fields fixed the problem.
In my opinion It would be a good idea to save also the git location and the python location within the plugin  and not in the workspace. The only thing that make sense to be saved in the workspace are which projects are currently part of the workspace (imported or created)

kolipakakondal, 2021-07-01:
I think that makes sense, I have provided a fix for this with https://github.com/espressif/idf-eclipse-plugin/pull/313. It will be available in the next release.

Thanks

kolipakakondal, 2021-08-19:
This is fixed in the v2.2.0 release, please get the latest changes using the update site.

closed 2021-08-19 by kolipakakondal