espressif/idf-eclipse-plugin #299: ESP-IDF Eclipse Plugin v2.1.0 on Eclipse 2021-06 cannot complete the install because of a conflicting dependency

bunjongy, 2021-06-21:
Cannot complete the install because of a conflicting dependency.
  Software being installed: Espressif IDF Plugins for Eclipse 2.1.1.202106040310 (com.espressif.idf.feature.feature.group 2.1.1.202106040310)
  Software currently installed: Eclipse IDE for C/C++ Developers 4.20.0.20210612-1200 (epp.package.cpp 4.20.0.20210612-1200)
  Only one of the following can be installed at once: 
    Simple Configurator 1.4.0.v20210315-2228 (org.eclipse.equinox.simpleconfigurator 1.4.0.v20210315-2228)
    Simple Configurator 1.3.600.v20200721-1308 (org.eclipse.equinox.simpleconfigurator 1.3.600.v20200721-1308)
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.1.1.202106040310 (com.espressif.idf.feature.feature.group 2.1.1.202106040310)
    To: org.eclipse.equinox.p2.iu; org.eclipse.equinox.simpleconfigurator [1.3.600.v20200721-1308,1.3.600.v20200721-1308]
  Cannot satisfy dependency:
    From: Eclipse IDE for C/C++ Developers 4.20.0.20210612-1200 (epp.package.cpp 4.20.0.20210612-1200)
    To: org.eclipse.equinox.p2.iu; org.eclipse.platform.feature.group [4.20.0.v20210611-1600,4.20.0.v20210611-1600]
  Cannot satisfy dependency:
    From: Eclipse 4 Rich Client Platform 4.20.0.v20210602-2209 (org.eclipse.e4.rcp.feature.group 4.20.0.v20210602-2209)
    To: org.eclipse.equinox.p2.iu; org.eclipse.equinox.simpleconfigurator [1.4.0.v20210315-2228,1.4.0.v20210315-2228]
  Cannot satisfy dependency:
    From: Eclipse Platform 4.20.0.v20210611-1600 (org.eclipse.platform.feature.group 4.20.0.v20210611-1600)
    To: org.eclipse.equinox.p2.iu; org.eclipse.rcp.feature.group [4.20.0.v20210611-1600,4.20.0.v20210611-1600]
  Cannot satisfy dependency:
    From: Eclipse RCP 4.20.0.v20210611-1600 (org.eclipse.rcp.feature.group 4.20.0.v20210611-1600)
    To: org.eclipse.equinox.p2.iu; org.eclipse.e4.rcp.feature.group [4.20.0.v20210602-2209,4.20.0.v20210602-2209]

kolipakakondal, 2021-06-21:
Hi @bunjongy 

Thanks for reporting this issue, it seems to be this is happening with the latest Eclipse release `2021-06` version with idf plugin, could you please use `Eclipse 2021-03` until we fix this issue.

Thanks

PascalBod, 2021-06-21:
I got the same problem. Switching back to Eclipse 2021-03...

gabrielfonte, 2021-06-22:
Same problem here

kolipakakondal, 2021-06-24:
Hi All, ESP-IDF Eclipse Plugin v2.1.2 (Build: 2.1.2.202106231127) is released to support Eclipse 2021-06(4.20) version. For update please check this https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.2

closed 2021-06-24 by kolipakakondal