espressif/idf-eclipse-plugin #314: Closing Eclipse With Project Open in Debug Profile Causes Subsequent Builds to Fail in Debug Profile EXP32-C3

BrianAtDocumentedDesigns, 2021-07-04:
**Describe the bug**

If you create a debug profile and properly configure it you can modify, compile, and JTAG debug an ESP32-C3. If you then close and re-open Eclipse, builds will fail (see https://github.com/espressif/idf-eclipse-plugin/issues/309#issuecomment-873046961). 

A workaround is to delete **ALL** debug profiles and create a new one from scratch. 

**To Reproduce**
Steps to reproduce the behavior:
This is Eclipse 2021-06 CDT, Eclipse Plugin V2.1.0, on Windows.

1. Create an ESP32 project, import blink. Configure as ESP32-C3
2. Create a debug profile (see https://www.esp32.com/viewtopic.php?f=40&t=21809 for how I do that)
3. Compile and flash to make sure it works
4. Exit Eclipse
5. Open Eclipse. Select the Debug Profile you just made. Note how the Upper Left Toolbar no longer provides the on: ESP32-C3 box (as per https://github.com/espressif/idf-eclipse-plugin/issues/309#issuecomment-873046961). Try and compile. It fails.
6. Delete all debug profiles and create a new one as per https://www.esp32.com/viewtopic.php?f=40&t=21809 
7. Now you can compile, etc., until you close Eclipse again.
 

**Expected behavior**
Closing Eclipse should not necessitate deleting debug profiles.

**Screenshots**

See https://github.com/espressif/idf-eclipse-plugin/issues/309#issuecomment-873046961

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Executing C:\Users\bjpic\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe C:\Users\bjpic\Desktop\Eclipse-ESP\esp-idf\tools\idf_tools.py list
WARNING: tool openocd-esp32 version v0.10.0-esp32-20210401 is installed, but has reported version v0.10.0-esp32-20210401-51-g84ee1302
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r1-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r1-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r1-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r1-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20210401 (recommended)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
[Eclipse Config.txt](https://github.com/espressif/idf-eclipse-plugin/files/6759195/Eclipse.Config.txt)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
2.1.0

**Eclipse Error log:** 
Window > Show View > Error Log
Attached. Also

Working Cmake
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\bjpic\Desktop\esp-idf\tools\cmake\toolchain-esp32c3.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32c3 C:\Users\bjpic\eclipse-workspace\blink

Failing Cmake
cmake -G Ninja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 C:\Users\bjpic\eclipse-workspace\blink4

[Eclipse Error Log.log](https://github.com/espressif/idf-eclipse-plugin/files/6759200/Eclipse.Error.Log.log)
[Eclipse Config.txt](https://github.com/espressif/idf-eclipse-plugin/files/6759201/Eclipse.Config.txt)

sigmaaa, 2021-07-15:
Hi @BrianAtDocumentedDesigns, thanks for reporting the issue. Fix for it and for the indexer issue on change configuration is available on the latest master. It's not included in the stable release yet. 

If you want to try the build with the fix, pick this: https://github.com/espressif/idf-eclipse-plugin/suites/3224044668/artifacts/74680530

And install as per the instructions here https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

BrianAtDocumentedDesigns, 2021-07-15:
Dear Sigmaaa
Thank you for the fix. Unfortuneatly I can't install the plugin. It fails with 

An error occurred during the org.eclipse.equinox.internal.p2.engine.phases.CheckTrust phase.
session context was:(profile=C__cpp-2021-06_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.CheckTrust, operand=, action=).
Error reading signed content.
C:\Users\bjpic\.p2\pool\features\ilg.gnumcueclipse.debug.gdbjtag.openocd.feature_4.4.1.202007271621 (The system cannot find the file specified)

My system has files like C:\Users\bjpic\.p2\pool\plugins\org.eclipse.embedcdt.debug.gdbjtag.openocd.core_5.0.0.202102181132.jar but not openocd.feature.

I don't know much about Eclipse plugins so I don't know what to do. 

I should mention that I am not using the ESP-IDF plugin in the distribution or master branches as it doesn't work with ESP32-C3. I am using a June version which does. (Though I don't think that is the problem).

Thanks
Brian

sigmaaa, 2021-07-20:
Hi @BrianAtDocumentedDesigns,

Try this build https://github.com/espressif/idf-eclipse-plugin/suites/3237375783/artifacts/75106304. It has fixes for issues, that I mentioned above, and also it contains a fix, that allows plugin work with ESP32-C3. 

I just tried to install it on the fresh Eclipse 2021-06 CDT on windows and it works fine. 
ilg.gnumcueclipse.debug.gdbjtag.openocd.feature_4.4.1.202007271621 this file should be provided by plugin. You can check this in the eclipse Help -> Install New Software... -> work with (select espressif plugin here) -> and check  Embedded C/C++ OpenOCD Debugging (incubation)
[image]

BrianAtDocumentedDesigns, 2021-07-20:
I managed to install the plugin by deleting Eclipse and the .p2 directory. Then I reinstalled Eclipse and then the plugin.

Unfortunately, OpenOCD no longer worked with the ESP32-C3. I updated the master branch of ESP-IDF and now absolutely nothing works. I am going to try to determine if it works in ESP-IDF.

It is very hard to solve a problem when there appear to be 3 independent developers (ESP-IDF, Eclipse Plugin, OpenOCD-ESP) and it takes many days between posting a problem/bug and having something work.

In general I'd say that support for ESP32-C3 is very poor.

BrianAtDocumentedDesigns, 2021-07-21:
Update:

I was able to get things to work, more or less, by completely wiping and re-installed ESP-IDF master branch.

I still have to delete the build directory if I change from run to debug configuration.

I noticed there are changed to the Debug Configuration -> Debugger tab
1) Although -f board/esp32c3-builtin.cfg is "remembered" the Board: setting is always blank
2) Actual executable should be C:\Users\bjpic\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe (the default is wrong)
3) When editing the debug configuration from the Eclipse bar the right scroll is missing so you don't notice the various options
4) Commands should be set mem inaccessible-by-default off set arch riscv:rv32

I suspect this is a problem with OpenOCD but flashing fails in debug

Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
** Flashing Failed **
-1

As such, you have to compile, switch to run configuration, run (which flashes the device (but also introduces all the unresolved includes), switch to debug, then debug. Note further that if you are using USB for console output you have to close your terminal emulator (or you can't program) but if you run debug without a terminal emulator, the debugger will appear to hang.

I am going to see if the version of OpenOCD from June which interprets the program script will work.

sigmaaa, 2021-07-21:
Hi, @BrianAtDocumentedDesigns,

flashing via jtag is now by default before starting the debug session. The OpenOCD version, which comes with the latest esp-idf, doesn't support the jtag flashing for esp32c3. The good news is that [the new openOCD](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20210721) version has just been released and it supports jtag flashing. 

You can check [this guide ]( https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/JTAG%20Flashing.md#providing-the-right-path-for-openocd)to install openOCD and use it in the plugin.

After editing `OPENOCD_SCRIPTS` variable, restart eclipse and global OpenOCD Path should be with the new provided version.
To check the global OpenOCD Path go to `Prefrences > Espressif > Global OpenOCD Path`

BrianAtDocumentedDesigns, 2021-07-23:
Sigmaaa

Everything now appears to work with one minor exception:

Creating a debug profile for ESP32-C3 autofills the wrong gbd. This field should point to 

riscv32-esp-elf-gdb.exe

I will raise another issue regarding this so you can keep track.

Thanks!

closed 2021-07-23 by BrianAtDocumentedDesigns