espressif/idf-eclipse-plugin #315: SDK Configuration editor do not recognize Kconfig "menuconfig ...." line and it parameters.

Maksons, 2021-07-06:
**Bug description**

Could not enable component or parameter because SDK configuration editor do not recognize Kconfig "menuconfig ..." definition.
How I  found it - tried to include libesphttpd component but could not enable it throgh GUI SDK configuration editor.

**To Reproduce**
1. Create new component with existing Kconfig file.
2. In Kconfig file put:
```
menu "mycomponent"

menuconfig MYCOMPONENT_ENABLED
    bool "mycomponent"
    help
        This enable is not visible in SDK configuration editor so could not enable component.

config MYCOMPONENT_PARAM
        int "Parameter 1"
        depends on MYCOMPONENT_ENABLED
        range 0 10
        default 1
        help
                MyComponent parameter which is not visible because component could not be enabled.

endmenu
```
3. Opend "sdkconfig" with SDK Configuration editor and try to find / enable new created component, there is no entry for "mycomponent" nor "submenu Parameter 1"

**Expected behavior**
Component tree entry "mycomponent" with enable checkbox.
Expected behavior could be acheved by replacing Kconfig line:
    menuconfig MYCOMPONENT_ENABLED
with 
    config MYCOMPONENT_ENABLED

**Environment Information:** 
Operating System: linux
Java Runtime Version: 11.0.11+9-Ubuntu-0ubuntu2.20.04
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF v4.4-dev-1404-gc13afea63-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.10
Python declared in PATH variable: Python 2.7.18

Maksons, 2021-07-14:
#### Problem still exist with updated plugin version.

Operating System: linux
Java Runtime Version: 11.0.11+9-Ubuntu-0ubuntu2.20.04
Eclipse Version: 4.20.0.v20210611-1600
Eclipse CDT Version: 10.3.0.202104042017
IDF Eclipse Plugin Version: 2.1.2.202106231127
ESP-IDF v4.4-dev-1404-gc13afea63-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.10
Python declared in PATH variable: Python 2.7.18

ShenRen, 2021-10-15:
我也提交了相同的问题，希望这些发现能够帮助你解决问题。
I submitted the same question in the hope that these findings will help you solve it.

SDK Configuration does not support "Depends on" control for menu display, but supports" visible if ".

https://github.com/espressif/idf-eclipse-plugin/issues/375

kolipakakondal, 2022-02-14:
Hello All, Fix is available as part of the IEP 2.4.0 release. Please get the latest changes using the update site to verify. Thank you.

closed 2022-02-14 by kolipakakondal