espressif/idf-eclipse-plugin #322: Can't type any input in a terminal when Open a Terminal with "ESP-IDF Serial Monitor" to connect to an ESP32 board

erusyd, 2021-07-20:
**Describe the bug**
In Eclispe CDT 2021-6 under Ubuntu 20.04, when 'Open a Terminal' and select 'ESP-IDF Serial Monitor' to connect to an ESP32-PICO-Kit board, I can't type any command in the terminal.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Open a Termianl' -> Choose terminal -> select 'ESP-IDF Serial Monitor'
2. Select proper project in 'Project name'
3. Select correct 'Serial port'
4. Click OK

**Screenshots**
[image]

**Environment Information:** 
Ubuntu 20.04 x86-64, Eclipse Embedded CDT 2021-6, openjdk 16.0.1 2021-04-20, Python 3.8.10, Espressif IDF Plugin for Eclipse uses Stable releases: https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

**ESP-IDF Tools Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - 1.24.0.123_64eb9ff-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	2.1.2.202106231127	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

kolipakakondal, 2022-02-14:
Hello, This issue got fixed with the IEP v2.4.0 release, could you please get the latest changes and verify. Thank you.

closed 2022-02-14 by kolipakakondal

erusyd, 2022-02-15:
> Hello, This issue got fixed with the IEP v2.4.0 release, could you please get the latest changes and verify. Thank you.

I verified it in the newly released Espressif IDE. Thanks for your guys' work!