espressif/idf-eclipse-plugin #325: Debug Configuration for ESP32-C3 wrong gdb

BrianAtDocumentedDesigns, 2021-07-23:
**Describe the bug**
After installing the eclipse plugin, creating a debug profile points to the wrong gdb executable.
 Actual Executable is 
<user>\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
it should be 
<user>\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Run -> Debug Configurations Select ESP-IDF GDB OpenOCD Debugging
2. Click on New Launch Configuration
3. Scroll down to Actual Executable
4. Note wrong gdb executable

**Expected behavior**
The field should point to riscv32-esp-elf-gdb.exe
The wrong gdb result in profuse errors

**Screenshots**

**Environment Information:** 

*** Features:

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
  Espressif IDF Plugins for Eclipse	2.1.2.202107141350	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

sigmaaa, 2021-07-26:
Hi @BrianAtDocumentedDesigns, 

fix for this is available in the latest master and in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/3301595730/artifacts/77183061):
To install, follow the instructions [here](https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile) .

Best regards,
Denys

BrianAtDocumentedDesigns, 2021-07-26:
Perhaps I am doing it wrong but I uninstalled the Espressif plugin downloaded the link you provided and install that and I get the same result

Actual executable 
C:\Users\bjpic\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gdb.exe

The file which downloads is 
https://github.com/espressif/idf-eclipse-plugin/suites/3301595730/artifacts/77183061

and is 15,100 kb, which is a different size from the previous ones. When I install 

  Espressif IDF Plugins for Eclipse	2.1.2.202107220757	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

The code 20210722 is the same date as the previous plugin I installed (I think)

Note that if you have an existing proper debug configuration, new debug configurations seem to inherit the actual executable from the existing one. Having the wrong gbd would mostly be a problem for people getting started.

JianWang18, 2021-08-13:
> Perhaps I am doing it wrong but I uninstalled the Espressif plugin downloaded the link you provided and install that and I get the same result
>
> Actual executable [...]

Has the problem been solved?

BrianAtDocumentedDesigns, 2021-08-13:
The above reply was 21 days ago so I tried a few times then I decided that I would stop trying to debug the tools. I move between problems with the plugin, openocd, and the documentation.

Perhaps at some point in the future I will try again but at the moment I am not focusing on debugging ESP32-C3 support.

kolipakakondal, 2021-08-14:
v2.2.0 has a fix for this. Please check https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0

closed 2021-08-14 by kolipakakondal

JianWang18, 2021-08-14:
> v2.2.0 has a fix for this. Please check https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0

The changlog show ESP-IDF Eclipse Plugin v2.2.0 (Build: 2.2.0.202108131352)
But Eclipse show:
IDF Eclipse 插件版本： 2.2.0.202108110455

sigmaaa, 2021-11-18:
Hi @BrianAtDocumentedDesigns, @JianWang18 
we are changing the gdb executable dynamically now, based on selected target. You can find these changes in the [2.3 release](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.3.0). So for 2.3, the gdb executable must be correct for sure. Thanks