espressif/idf-eclipse-plugin #326: The following Python requirements are not satisfied: python-socketio<5

mohannadmaklad, 2021-07-24:
I have installed IDF Tools using offline installer (Full Installation), and I can build without issues using idf.py build
But when I try to compile with eclipse, I get this error message : 
The following Python requirements are not satisfied:
python-socketio<5

Full console output : 

Building in: C:\Users\Mohannad\eclipse-workspace\kk\build
Configuring in: C:\Users\Mohannad\eclipse-workspace\kk\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\Mohannad\OneDrive\Desktop\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 C:\Users\Mohannad\eclipse-workspace\kk
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.30.0.windows.2") 
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Mohannad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Mohannad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Mohannad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Mohannad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Mohannad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
python-socketio<5
To install the missing packages, please run "C:\Users\Mohannad\OneDrive\Desktop\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\Mohannad\.espressif\python_env\idf4.4_py3.8_env
    Python interpreter used: C:\Users\Mohannad\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
CMake Error at C:/Users/Mohannad/OneDrive/Desktop/esp-idf/tools/cmake/build.cmake:274 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/Mohannad/OneDrive/Desktop/esp-idf/tools/cmake/build.cmake:400 (__build_check_python)
  C:/Users/Mohannad/OneDrive/Desktop/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Mohannad/eclipse-workspace/kk/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.

Build complete (0 errors, 0 warnings): C:\Users\Mohannad\eclipse-workspace\kk\build
Total time taken to build the project: 4,068 ms

Photo for a better readability : 
[image]

sigmaaa, 2021-07-26:
Hi @mohannadmaklad,

try to install tools via eclipse.  Help > ESP-IDF Tools Manager > Install tools
It should check dependencies and installed the missed ones.

georgik, 2021-07-26:
@mohannadmaklad Thank you for reporting the problem. 
The IDF version in the report is 4.4. The Offline installer provides only 4.3 and 4.2.1 https://github.com/espressif/idf-installer#table-with-downloads
The ESP-IDF must came from other source so the package with socket is missing.

To fix the problem:
- download small online installer e.g. 2.9: https://dl.espressif.com/dl/esp-idf/
- start the installation, in the third screen choose `Use existing ESP-IDF directory`, click browse and select the directory
- perform the installation, it should be fast, because all files are already in place

Other option:
- open terminal with ESP-IDF and run install.bat command
- the missing dependency should be downloaded

BTW: I'm curious about installation on OneDrive. Does it work and is it able to sync across multiple computers? What is the performance? It might be an interesting option for Windows users.

mohannadmaklad, 2021-07-26:
@georgik For OneDrive, it's not intended at all, it just shows up when I install IDF @ my desktop, never had your thoughts :D 

@sigmaaa @georgik  Actually I have tried all of these, but I think the problem is that i already had the IDF installed, then decided to try the offline installer so things got messed up.

I had a dirty solution buy commenting python-socketio<5 in the text file and it works well.

georgik, 2021-07-27:
@mohannadmaklad Thank you for providing details. I'll add testing scenario to Offline installer. The environment should be separate and they should not influence each other.

Let me add bit of theory to explain how to fix it.

Each minor version of ESP-IDF has it's own Python Virtual Environment to keep them isolated, because they have slightly different prerequisites. You can find the environment in `.espressif/idf4.3_py3.8_env`. The version numbers corresponds to IDF version and Python version. The environment gets activated when you launch ESP-IDF from desktop or simply type `./export.bat` in the directory of your ESP-IDF.

It means that 4.3 and 4.4 can live safely side-by-side, because they have different directories for packages of virtualenv. The error message that you see indicates that in the environment 4.4 is missing one package. The export.bat or install.bat.

You can call the virtualenv Python directly and tell it to install missing package:
```
C:\Users\Mohannad\.espressif\python_env\idf4.4_py3.8env\Scripts\python.exe -m pip install "python-socketio<5"
```

You can list installed packages :
```
C:\Users\Mohannad\.espressif\python_env\idf4.4_py3.8env\Scripts\python.exe -m pip list
```

georgik, 2021-07-28:
@mohannadmaklad Meanwhile I was able to simulate the problem with scenario:
- install 4.3 by offline installer
- update git to master

The issue and workaround is described here: https://github.com/espressif/idf-installer/issues/38

mohannadmaklad, 2021-07-28:
@georgik  Thanks georgik! I will close the issue.

closed 2021-07-28 by mohannadmaklad

georgik, 2021-07-29:
I'd like to add correct procedure when upgrading between minor versions of IDF. E.g. from 4.3 to 4.4:
- run `install.bat`
- run: `git submodule update --init --recursive`
- close the terminal
- open new terminal

Opening the new terminal is important, because export.bat is not able to handle change of already exported variables from previous version. It's necessary to start the process from the scratch.