espressif/idf-eclipse-plugin #328: Error during building the project

Dejwuch, 2021-07-26:
Hello, 
I was trying to solve this problem by myself but I don't make it.. Please let me help.
I have installed ESP-IDF (v4.3) and Eclipse (2021-03) via ESP-IDF Tools 2.9. I was doing this according to [https://github.com/espressif/idf-eclipse-plugin](https://github.com/espressif/idf-eclipse-plugin)

All the time the error is the same:

```
Building in: E:\Dawid\Programowanie\Eclipse\hello_world\build
Configuring in: E:\Dawid\Programowanie\Eclipse\hello_world\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\esp-idf\esp-idf_v4.3\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 E:\Dawid\Programowanie\Eclipse\hello_world
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.30.1.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Dawid/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Dawid/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Dawid/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Dawid/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Dawid/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
CMake Error at C:/esp-idf/esp-idf_v4.3/tools/cmake/build.cmake:274 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/esp-idf/esp-idf_v4.3/tools/cmake/build.cmake:400 (__build_check_python)
  C:/esp-idf/esp-idf_v4.3/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Dawid/Programowanie/Eclipse/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The specified file could not be found.

Build complete (0 errors, 0 warnings): E:\Dawid\Programowanie\Eclipse\hello_world\build
Total time taken to build the project: 7 834 ms
```

I attached also CMakeOutput.log 
[CMakeOutput.log](https://github.com/espressif/idf-eclipse-plugin/files/6880909/CMakeOutput.log)

**Expected behavior**
Build the example project from templates: "Hello_world" without errors.

**Environment Information:** 
[Environment Information.txt](https://github.com/espressif/idf-eclipse-plugin/files/6880875/Environment.Information.txt)

**ESP-IDF Tools Information:**
```
Executing C:\Users\Dawid\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe C:\esp-idf\esp-idf_v4.3\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - 1.24.0.123_64eb9ff-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
[... 92 lines cut ...]
Requirement already satisfied: greenlet>=0.4.14 in c:\users\dawid\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gevent<2.0,>=1.2.2->gdbgui==0.13.2.0->-r C:\esp-idf\esp-idf_v4.3\requirements.txt (line 15)) (1.1.0)
Requirement already satisfied: MarkupSafe>=2.0 in c:\users\dawid\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from Jinja2>=2.4->Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r C:\esp-idf\esp-idf_v4.3\requirements.txt (line 15)) (2.0.1)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py): started
  Building wheel for esp-windows-curses (setup.py): finished with status 'done'
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-py3-none-any.whl size=1171 sha256=0ccc7dbb47040a04fc24a7745d6d1abc0b23331928fc442fc947f7fb4a6f2659
  Stored in directory: c:\users\dawid\appdata\local\pip\cache\wheels\94\ad\e8\4b2641f7da378dab169e1510606378c76a0a2c53ff7ef577e4
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1
Installing Python packages from C:\esp-idf\esp-idf_v4.3\requirements.txt

Executing C:\Users\Dawid\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe C:\esp-idf\esp-idf_v4.3\tools\idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=C:\Users\Dawid\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE=1
IDF_PYTHON_ENV_PATH=C:\Users\Dawid\.espressif\python_env\idf4.3_py3.9_env
PATH=C:\Users\Dawid\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\Dawid\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Dawid\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\Dawid\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\Users\Dawid\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Dawid\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Dawid\.espressif\tools\cmake\3.16.4\bin;C:\Users\Dawid\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;C:\Users\Dawid\.espressif\tools\ninja\1.10.2\;C:\Users\Dawid\.espressif\tools\idf-exe\1.0.1\;C:\Users\Dawid\.espressif\tools\ccache\3.7\;C:\Users\Dawid\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\Dawid\.espressif\python_env\idf4.3_py3.9_env\Scripts;C:\esp-idf\esp-idf_v4.3\tools;%PATH%
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.
```

franz-ms-muc, 2021-08-23:
have exactly the same Problem: 

Building in: C:\git\hr\CTIS2\HR-Agrar-ISO-OS\build
Configuring in: C:\git\hr\CTIS2\HR-Agrar-ISO-OS\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\FranzHoepfinger\esp-idf-v4.3\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 C:\git\hr\CTIS2\HR-Agrar-ISO-OS
-- Found Git: C:/Users/FranzHoepfinger/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/FranzHoepfinger/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/FranzHoepfinger/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/FranzHoepfinger/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/FranzHoepfinger/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/FranzHoepfinger/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=7.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
construct==2.10.54
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\FranzHoepfinger\esp-idf-v4.3\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\FranzHoepfinger\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: C:\Users\FranzHoepfinger\.espressif\tools\idf-python\3.8.7\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\Users\FranzHoepfinger\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\FranzHoepfinger\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\FranzHoepfinger\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\FranzHoepfinger\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\Users\FranzHoepfinger\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\FranzHoepfinger\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\FranzHoepfinger\.espressif\tools\cmake\3.16.4\bin;C:\Users\FranzHoepfinger\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;C:\Users\FranzHoepfinger\.espressif\tools\ninja\1.10.2\;C:\Users\FranzHoepfinger\.espressif\tools\idf-exe\1.0.1\;C:\Users\FranzHoepfinger\.espressif\tools\ccache\3.7\;C:\Users\FranzHoepfinger\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\FranzHoepfinger\esp-idf-v4.3\tools;C:/Users/FranzHoepfinger/.espressif/tools/idf-eclipse/2021-07//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_16.0.1.v20210528-1205/jre/bin/server;C:/Users/FranzHoepfinger/.espressif/tools/idf-eclipse/2021-07//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_16.0.1.v20210528-1205/jre/bin;C:\Users\FranzHoepfinger\.espressif\tools\idf-python\3.8.7\;C:\Users\FranzHoepfinger\.espressif\tools\idf-git\2.30.1\cmd\;C:\Program Files\Google\Chrome\Application;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Crucial\Crucial Storage Executive;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Users\FranzHoepfinger\.espressif\tools\idf-git\2.30.1\cmd;C:\Program Files\nodejs\;C:\Program Files\Google\Google Apps Sync\;C:\Program Files\Google\Google Apps Migration\;C:\Tools\PoolValidationTool;C:\Users\FranzHoepfinger\.espressif\python_env\idf4.3_py3.8_env\Scripts;C:\Users\FranzHoepfinger\AppData\Local\Microsoft\WindowsApps;C:\Users\FranzHoepfinger\AppData\Roaming\npm;C:\Users\FranzHoepfinger\AppData\Local\gitkraken\bin;C:\Infineon\Toolbox Launcher;C:\Infineon\LauncherService;C:\Users\FranzHoepfinger\esp-idf-v4.3
CMake Error at C:/Users/FranzHoepfinger/esp-idf-v4.3/tools/cmake/build.cmake:274 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/FranzHoepfinger/esp-idf-v4.3/tools/cmake/build.cmake:400 (__build_check_python)
  C:/Users/FranzHoepfinger/esp-idf-v4.3/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "C:/git/hr/CTIS2/HR-Agrar-ISO-OS/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': Das System kann die angegebene Datei nicht finden.

Build complete (0 errors, 0 warnings): C:\git\hr\CTIS2\HR-Agrar-ISO-OS\build
Total time taken to build the project: 3.605 ms

franz-ms-muc, 2021-08-23:
this seems curious: 

Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\FranzHoepfinger\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: C:\Users\FranzHoepfinger\.espressif\tools\idf-python\3.8.7\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH

franz-ms-muc, 2021-08-23:
idf.py build works FINE from the Console.

franz-ms-muc, 2021-08-23:
another thing: 

if 1 time built from Console `idf.py build`

then it works to build from Eclipse. 

Operating System: windows 10  
Java Runtime Version: 16.0.1+9-24  
Eclipse Version: 4.20.0.v20210611-1600  
Eclipse CDT Version: 10.3.0.202104042017  
IDF Eclipse Plugin Version: 2.2.0.202108110455

franz-ms-muc, 2021-08-23:
does this help anything: 

Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

franz-ms-muc, 2021-08-28:
it is solved with Release IDF Eclipse Plugin Version: 2.2.0.202108110455

sigmaaa, 2021-08-30:
Hi @franz-ms-muc, @Dejwuch 
thanks for reporting the problem. I can also confirm, that I wasn't able to reproduce the problem on the latest released version. I'm closing the issue for now. Feel free to re-open if you will face this problem again. Thanks

closed 2021-08-30 by sigmaaa

dizcza, 2021-09-07:
I do face the same issue with the latest Eclipse IDF plugin.
I've described the issue [here](https://github.com/espressif/idf-eclipse-plugin/issues/228#issuecomment-914040016).
Manual building with the command line does not help though.