espressif/idf-eclipse-plugin #346: Eclipse on Windows shows error markers for cpp file but not for c file when I include freertos/FreeRTOS.h but build succeeds anyway (IDFGH-5757)

davidlehrian, 2021-08-25:
## Environment
- Development Kit:      ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.4-dev-2594-ga20df743f
- Build System: CMake 3.20.3
- Compiler version: esp-2021r1-8.4.0
- Operating System:  Windows 10
- Using an IDE?: Yes Eclipse 
  Operating System: windows 10
  Java Runtime Version: 15.0.1+9
  Eclipse Version: 4.20.0.v20210611-1600
  Eclipse CDT Version: 10.3.0.202104042017
  IDF Eclipse Plugin Version: 2.2.0.202108110455
  ESP-IDF v4.4-dev-2594-ga20df743f
  Python set for IDF_PYTHON_ENV: Python 3.8.10
  Python declared in PATH variable: Python 3.9.5
- Power Supply:   USB
- 
## Problem Description

Eclipse IDE is showing an error marker in a cpp file but the same code in a c file does not produce the error.  Both files compile without error.  

### Expected Behavior

I should not be seeing an error marker in the cpp file.  

### Actual Behavior

I am seeing an error marker in the cpp file.  

### Steps to reproduce

1. Create a new project from the sample_project template.
2. Create the following test_function.c and test_function.h files.
```
/* this is a test file */

#include <esp_system.h>

void test_function_a();
```
```
#include <freertos/FreeRTOS.h>
#include <driver/gpio.h>
#include "test_function.h"

void test_function_a(){
	gpio_pad_select_gpio(26);
	gpio_set_direction(26, GPIO_MODE_OUTPUT);
	gpio_set_level(26,0);
}
```
3. Compile the project and notice a clean compile and there are no error markers in Eclipse IDE.
4. Create the following TestClass.cpp and TestClass.h files

```
/* test class */
#include <esp_system.h>

#ifndef CLASS_TestClass_H_
#define CLASS_TestClass_H_

class TestClass {
private:
	TestClass();
	uint8_t GPIO;
public:
	TestClass(uint8_t pin);
	void testMethod();
};

#endif
```
```
/* test class */
#include <freertos/FreeRTOS.h>
#include <driver/gpio.h>
#include "TestClass.h"

TestClass::TestClass(uint8_t pin){
	GPIO=pin;
}

void TestClass::testMethod(){
	gpio_pad_select_gpio((unsigned char)GPIO);
	gpio_set_direction((gpio_num_t)GPIO, GPIO_MODE_OUTPUT);
	gpio_set_level((gpio_num_t)GPIO,0);
}
```
5. Compile the project and notice a clean compile however Eclipse is marking lines 11, 12 and 13 with error markers.

[image]

6. Now comment out the the first include line in TestClass.cpp 
`//#include <freertos/FreeRTOS.h>`
7. Notice the error markers go away.

[image]

8. I have tried right clicking on the project and selecting Index-> Rebuild but it doesn't help.
9. I know the example isn't using anything from <freertos/FreeRTOS.h> but I was trying to use taskENTER_CRITICAL and taskEXIT_CRITICAL in another project and that is when I came across this problem.  This example is a simple example that recreates the problem without including all the other extraneous code from my project.

brianignacio5, 2021-08-25:
Are you using the `idf-eclipse-plugin` ? You can check the [README configure environment variables](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#configuring-environment-variables) ?

davidlehrian, 2021-08-25:
I am using the idf-eclipse-plugin and the environment variables were set up correctly by 

> Help > ESP-IDF Tools Manager > Install Tools

when I installed the system.  I am able to successfully compile and flash a device so there is no problem with my installation as far as I can tell.  I came across this error when I took a set of functions I had written, that work just fine as functions, and turned them into a class.  The class I created will compile and work just fine too.  The issue is that Eclipse seems to think there is an error when I include <freertos/FreeRTOS.h> in the class, but it doesn't when I include it in the function.  Here is the file with the functions in them.  

[image]

It seems there is something funny with how the idf include is processed by the indexer when included in a cpp class.

sigmaaa, 2021-08-31:
Hi @davidlehrian

what do you have in CMakeLists.txt which is in the main folder? 
try this:
```
set(srcs "main.c"
	"test_function.c"
	"TestClass.cpp"
	)

idf_component_register(SRCS ${srcs}
					INCLUDE_DIRS ".")
```

davidlehrian, 2021-08-31:
Thanks for the suggestion.  Here is my CMakeLists.txt

```
idf_component_register(SRCS "main.c" "test_function.c" "TestClass.cpp"
                    INCLUDE_DIRS ".")
```
Your CMakeLists.txt does not change the behavior of the markers in Eclipse, but I see what you did there and I will use your format because it looks cleaner ;-)

davidlehrian, 2021-08-31:
I was just playing around I discovered something new.  It doesn't seem to matter what file I include from freertos I get the same markers.  For example if I include <freertos/task.h> instead of <freertos/FreeRTOS.h> the error markers show up.  

```
//#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/gpio.h>
#include "TestClass.h"

TestClass::TestClass(uint8_t pin){
	GPIO=pin;
}

void TestClass::testMethod(){
	gpio_pad_select_gpio((unsigned char)GPIO);
	gpio_set_direction((gpio_num_t)GPIO, GPIO_MODE_OUTPUT);
	gpio_set_level((gpio_num_t)GPIO,0);
}
```

davidlehrian, 2021-10-06:
I was able to find a solution to this issue.  Not sure why this works, but if I move the include of gpio.h above FreeRTOS.h the error markers go away.  This fixes the problem.  

```
#include <driver/gpio.h>
#include <freertos/FreeRTOS.h>
#include "TestClass.h"

TestClass::TestClass(uint8_t pin){
	GPIO=pin;
}

void TestClass::testMethod(){
	gpio_pad_select_gpio((unsigned char)GPIO);
	gpio_set_direction((gpio_num_t)GPIO, GPIO_MODE_OUTPUT);
	gpio_set_level((gpio_num_t)GPIO,0);
}

```

Why does the order of the includes seem to make a difference?

davidlehrian, 2021-10-12:
Thanks for the update!

kolipakakondal, 2022-02-14:
Hello All, Fix is available as part of the IEP 2.4.0 release. Please get the latest changes using the update site to verify. Thank you.

closed 2022-02-14 by kolipakakondal

Richard-CRT, 2023-08-05:
I assume I might be doing something wrong, but I'm still seeing this exactly behaviour using `Espressif-IDE 2.10.0 with ESP-IDF 5.0.2` (installed today). Fixed by moving the `driver/gpio.h` include above `freertos/FreeRTOS.h`

kolipakakondal, 2023-12-01:
> I continue to have this behavior on Eclipse 2023-09 running with ESP-IDF 5.1.1 and version 2.11.1.202310270725 of the Espressif plugin. It's been very frustrating. I've wrestled with the indexer issues for a couple of years. In that time there have been one or two times the problem magically nearly disappeared but then would come back. This is all on the same project through its history. I've tried just reimporting the project from scratch and reinstalling the Eclipse tool with plugin but that didn't solve it.

Hi @mbratch Unfortantely Eclipse CDT Indexer is not actively maintained, doesn't support C/C++ 20 lang features but some of these features are used in the ESP-IDF 5.0 and higher versions, hence you see indexer parser issues in IDE.To avoid all these problemss we are recommending to use clangd indexer, please use [these](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md) instructions to confgure in the Espressif-IDE. Going forward, will set this as the default option in the Espressif-IDE.

mbratch, 2023-12-01:
> Hi @mbratch Unfortantely Eclipse CDT Indexer is not actively maintained, doesn't support C/C++ 20 lang features but some of these features are used in the ESP-IDF 5.0 and higher versions, hence you [...]

Hi @kolipakakondal , thank you for the quick reply. I was not familiar with this news. I will try the clangd indexer.

mbratch, 2023-12-02:
Hi @kolipakakondal. I followed the instructions carefully on that link. The installation of the clangd indexer went smoothly but it doesn't seem to do anything. It appeared to go through an indexing process but now I have no hover or right-click view of cross references or other indexer functions. There's no context help. The message I get in the lower status bar is that it can't find the symbol that I'm looking for.

Jaboop8, 2024-01-24:
> Hi @mbratch Unfortantely Eclipse CDT Indexer is not actively maintained, doesn't support C/C++ 20 lang features but some of these features are used in the ESP-IDF 5.0 and higher versions, hence you [...]

Hi @kolipakakondal, I am having similar indexing issues. I tried following the steps that you linked above but ran into issues trying to set the clangd path. I used the [ESP-IDF Windows Installer](https://dl.espressif.com/dl/esp-idf/) to install the latest stable build (ESP-IDF 5.1.2 with Espressif IDE 2.12.0). I then launched the Espressif IDE closed the welcome page and checked the Build Enviornment variables. For me they did not automatically configure maybe because this is not my only instance of ESP-IDF but I was able to manually add them by echoing the enviornment variables in the ESP-IDF 5.1 CMD and copying them into the Espressif IDE. I was then able to import a known good project build and add launch configurations. 

Unfortunately I am stuck at this point. I installed the latest version of Eclipse CDT-LSP and the dependencies from the update site but then when I tried to add the clangd path I am unable to find it anywhere on my computer. I do not have a '.espressif' folder and I have checked the Espressif5.1.2 folder where I installed the ESP-IDF. Here I am able to find the GCC compiler path from the instructions but no clangd. Any idea where I can find this or if I am missing something obvious. I've showed a screenshot that is claiming everything should already be downloaded and up to date.

[image]

kolipakakondal, 2024-01-24:
Hi @Jaboop8 It's part of the esp-clang toolchain, please check the screenshot below. I think this is available from the esp-idf 5.0 and higher versions.

[image]

Jaboop8, 2024-01-24:
@kolipakakondal That is where I was hoping to find it. I see all those in the tools folder and a few more with the ESP-IDF 5.1.2 but not the esp-clang toolchain. Am I able to download that from somewhere else?

[image]

davidlehrian, 2024-01-24:
@kolipakakondal I can confirm what @Jaboop8 is stating. When installing the Espressif IDE via the Offline Installer [at this link](https://dl.espressif.com/dl/esp-idf/) the installation does not create a .espressif folder under the user's account like when using the Eclipse IDE and installing the Espressif plugin, but instead appears to use the tools and frameworks folders underneath the installation folder for the IDE which by default on Windows is C:\Espressif. I can also confirm that there is no esp-clang in the tools folder in the installation. 

However, if you go to the Espressif IDE menu and choose ESP-IDF Tools Manager -> Install Tools it will install the esp-clang folder under C:\Espressif\tools and you can configure the IDE as described [via the link provided](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md).

Having done this I do get code completion for the ESP-IDF C language and the errors I was getting on ESP-IDF calls have gone away, but I am now getting errors on includes like <String> and <mutex> and when I attempt to run the program I get an error popup in the IDE stating there are errors and asking if I want to launch it anyway. I understand these are CPP libraries and we just installed a clangd file so it may not support cpp, but this fix seems to switch one set of clang errors for another set of cpp errors. 

Also I do not get any context color highlighting in my cpp or h files like I get in my straight Eclipse installation which may be due to a plugin not being installed in my Espressif IDE that I have in Eclipse IDE. I am going to look at this a little more closely and will provide an update.

kolipakakondal, 2024-01-25:
> @kolipakakondal I can confirm what @Jaboop8 is stating. When installing the Espressif IDE via the Offline Installer [at this link](https://dl.espressif.com/dl/esp-idf/) the installation does not [...]

Thanks for reporting this. Let me take a look. @Jaboop8 Can you try `ESP-IDF Tools Manager -> Install Tools` and see if you are able to see esp-clang tools.

> I am now getting errors on includes like <String> and <mutex>

@davidlehrian Did you set the clangd drivers path in the preferences? Please check this https://github.com/espressif/idf-eclipse-plugin/issues/861#issuecomment-1849309369

Jaboop8, 2024-01-25:
> Thanks for reporting this. Let me take a look. @Jaboop8 Can you try `ESP-IDF Tools Manager -> Install Tools` and see if you are able to see esp-clang tools.
> 

I am having trouble setting the paths to Install the tools. I've tried what is below in the screenshot and I have tried the ESP-IDF path as being 'C:\Espressif5.1.2\frameworks\esp-idf-v5.1.2' and 'C:\Espressif5.1.2' for both sets of Git and Python paths but am getting errors for both. What paths am I supposed to be using?

[image]
[image]

This is the error I am seeing. I do see a tools.json at this file at path: C:\Espressif5.1.2\frameworks\esp-idf-v5.1.2\tools\tools.json but not sure if this is the same file the install tools is trying to find.
[image]

I then tried using the tool installation wizard and this seems to have partially worked but ends up throwing the same error towards the end of the process. I now have a .espressif folder but no clang bin.

[image]

davidlehrian, 2024-01-25:
@Jaboop8 It looks like you don't have git installed. "WARNING: Git was not found". Looking at idf_tools.py it uses git during installation to check the version of the IDF hence the error. You can [download it from here](https://gitforwindows.org/). That is the version I have installed and it seems to work.

Jaboop8, 2024-01-29:
@kolipakakondal @davidlehrian 
I do have git installed but it was an older version so I went ahead and updated to the 2.43 version you linked. I still see the same issue though. It looks to me like the issue is happening because it is searching for non-existent paths. It seems like it thinks there should be a 'src' folder at this level that is not there.
[image]

I tried adding one and putting everything inside of 'src' but then I get a different error.
[image]

**Solved:**
I copied the folders below into a src folder and put them at the same folder level as the Espressif5.1.2 folder
[image]
[image]

Not sure why these files needed to be in both places but resolved the issue with the scripts looking for the paths like:
'../../../../src\\tools/tools.json'.

Jaboop8, 2024-01-29:
@davidlehrian I am using a new workspace for the 5.1.2 ESP-IDF version.

I think I am getting close to where you are at now. I have no syntax coloring and I can't use standard keyboard shortcuts like 'Ctrl+/' to comment things out. However, I am also seeing errors on a lot of the sample code mostly with undeclared structs and a lot of implicit declaration errors. 
This project I am testing builds when I am not using the clangd but obviously has the error marking issues this is supposed to be resolving. I do see that warnings are being treated as errors. Maybe this is the difference, if there is a way to turn this off for now I could test and see if I am at least successfully building again. Seems weird that the idf example projects would have warnings that are being treated as errors though so maybe I'm still missing something.
[image]

That does explain the path issue I was having. There was an old environment variable set with that IDF_PATH. It was being overwritten anytime I ran the ESP-IDF 5.1 CMD terminal so I wasn't seeing it when I checked the path in there.

davidlehrian, 2024-01-29:
Glad you figured out the path/environment variable issue. 

Regarding the warnings you are getting, this is the result of the Drivers path setting in Preferences->C/C++->Editor (LSP)->clangd. [The link provided](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/clangd_cdt_support.md) that explains how to set up clangd says to set the Drivers to C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc but this driver doesn't resolve C++ files only the IDF C files. There is a discussion about this in issue [#861](https://github.com/espressif/idf-eclipse-plugin/issues/861#issuecomment-1849309369). I have found that the best value to use for the Drivers setting on windows is:

C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-*.exe

This resolves C as well as C++ libraries and I can compile all my code and examples with warning.

Jaboop8, 2024-01-29:
@davidlehrian 
I saw your note on this. I am using the same Drivers path but I do still see errors.
[image]

davidlehrian, 2024-01-29:
Do you have the .clangd file in the root of your project folder? It needs to contain:
```
CompileFlags:
  Remove: [-fno-tree-switch-conversion, -fstrict-volatile-bitfields]
```
If this doesn't exist there will be an error at the first comment in the file.

Jaboop8, 2024-01-29:
@davidlehrian 
I have the .clangd file. I just tested some other example projects and they are compiling fine. It seems the one I was working with the issue is around websockets. It looks like 90% of the errors are being thrown around this 'httpd_ws_frame_t' struct that is not resolving. 

I have the websocket support enabled in the sdkconfig but maybe there is something else weird with websockets. The only place I see this referenced is in the 'esp_idf_components/esp_http_server/src/httpd_ws.c' but looking at this it seems that the struct is actually defined in the 'esp_http_server.h' which is the include not resolving in the 'httpd_ws.c'. Is this just a component registration issue?

[image]

I found the definition in the ESP-IDF v4.4.2 since it is easier to find things with the indexer enabled and verified its in the same spot in the v5.1.2 but not sure why it doesn't resolve maybe this is just some CMake issue I'm not understanding though and not actual clangd setup.
[image]

davidlehrian, 2024-01-29:
Hmmm, odd. I just created a new project from their template wss_server which is their HTTP server with web socket support for ESP32. It is located under protocols\https_server\wss_server. After adding the .clangd file to the root of the project, it compiled without warning or error. I was able to start typing httpd_ws_frame_t and clangd found it and I was able to create a variable of this structure type. Can you try creating and compiling this example?

Jaboop8, 2024-01-30:
This is working for me too starting with a fresh example.

I guess the thing I have to figure out now is why the Import tool isn't working to bring in a project that was from v4.4.2 and get it to compile in v5.1.2. I took a look through the project settings and saw the linked sources were doubled up with both the v4.4.2 and v5.1.2 components so I went ahead and deleted the v4.4.2 linked references but am still seeing issues.
[image]

@kolipakakondal Any advice on importing projects from older ESP-IDF versions?

The other common error I am seeing is with sprintf formatters. Sort of weird too I see a lot of these errors but looking at the example below uint32_t is being labeled as 'long unsigned int' in the error message but as 'unsigned int' by clangd in the editor so no matter which formatter I use '%u' or '%lu' I am getting an error message for both. Also when I try to use the '%lu' formatter like the error message suggests I get a recommended fix to change it back.
[image]
[image]
[image]

davidlehrian, 2024-01-30:
Upgrading from 4 to 5 I was not able to import the project directly. What I did was create a new workspace and check out my source code from the repository and then import the project from the source. 

I think you might be able to get away with deleting the build folder and the .project and .cproject files from the directory, deleting the project from the workspace and then importing it. The project definitely has to be deleted from the IDE as I recall 4 created some sort of linked folders in the IDE to the ESP-IDF files which isn't done any more. This folder didn't exist on disk only in the IDE. I think doing this would also clear up the linked references to which you are referring.