espressif/idf-eclipse-plugin #352: Reducing verbosity  in build

icodk, 2021-08-30:
**Describe the issue**
When building a project, the CDT Global Build Console  is filled with text, and it is sometimes difficult to find warnings and errors. So most of the time I would like to keep it to the minimum.
How can I control it ?

At the moment, I get something similar to: 

`Building in: C:\Dev\Solar_power\eclipse\controlPanel\build
cmake --build . -- -v
[1/8] ccache C:\Dev\esp32-idf\idf-tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -IC:/Dev/esp32-idf/esp-idf-v4.2/components/newlib/platform_include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/freertos/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/freertos/xtensa/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/heap/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/log/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/lwip/include/apps -IC:/Dev/esp32-idf/esp-idf-v4.2/components/lwip/include/apps/sntp -IC:/Dev/esp32-idf/esp-idf-v4.2/components/lwip/lwip/src/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/lwip/port/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/lwip/port/esp32/include/arch -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/src/esp32/. -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/src/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_rom/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_common/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_system/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/xtensa/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/xtensa/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/driver/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/driver/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_ringbuf/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/efuse/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/efuse/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/espcoredump/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_timer/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_ipc/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/soc/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/soc/esp32/../include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/soc/soc/esp32/private_include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/vfs/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_wifi/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_wifi/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_event/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_netif/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/esp_eth/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/tcpip_adapter/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/app_trace/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/mbedtls/port/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/mbedtls/mbedtls/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/mbedtls/esp_crt_bundle/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/bootloader_support/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/app_update/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/spi_flash/include -IC:/Dev/esp32-idf/esp-idf-v4.2/components/wpa_supplicant/include -IC:/Dev/esp32- ...`

kolipakakondal, 2021-08-31:
Hi @icodk 

You could pass additional arguments to the CMake build to control the verbose output. https://cmake.org/cmake/help/v3.18/envvar/VERBOSE.html
https://github.com/espressif/idf-eclipse-plugin#launch-configuration

Open launch configuration > Navigate to Build settings tab > Provide additional CMake arguments

Please also check https://cmake.org/cmake/help/v3.19/command/target_compile_options.html?highlight=target_compile_options

target_compile_options can be defined in the CMakeLists.txt

icodk, 2021-08-31:
@kolipakakondal: I Tried both, none worked for me. Probably missing something:
1. I added the CMAKE_NO_VERBOSE to the Additional CMake augments in the Build settings tab of the Launch configuration
This didn't change anything in the build output.
2. I added the following line to the CMakeLists.txt file located in the root directory of the project:
`target_compile_options(controlPanel PUBLIC "CMAKE_NO_VERBOSE")`
and got the following error:
```
CMake Error at CMakeLists.txt:10 (target_compile_options):
Cannot specify compile options for target "controlPanel" which is not built
  by this project.
```
The CMakeLists.txt with the extra line I added looks like:
```
cmake_minimum_required(VERSION 3.5)

if (NOT DEFINED PROJECT_NAME)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
    project(controlPanel)
else()
    message(FATAL_ERROR "LV PORT ESP32: This must be a project's main CMakeLists.txt.")
endif()

target_compile_options(controlPanel PUBLIC "CMAKE_NO_VERBOSE")
```
What am I missing ?

kolipakakondal, 2021-09-01:
Hi @icodk 

I was talking to our esp-idf colleagues on this, as per them there is no IDF build system support for this.

Thanks

icodk, 2021-09-01:
@kolipakakondal 
Thanks

closed 2021-09-01 by icodk

aussiemate32, 2022-09-14:
cmake --build . -- -v.  why is make called with -v?