espressif/idf-eclipse-plugin #353: Hi guys ,

ssravindhar, 2021-09-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

kolipakakondal, 2021-09-01:
Hello @ssravindhar 

Please take a look at this example https://github.com/espressif/esp-aws-iot/tree/release/beta/examples/ota/ota_mqtt

This recent blog post might be helpful: https://blog.espressif.com/support-for-lts-release-of-aws-iot-device-sdk-for-embedded-c-on-esp32-8eeeea28b79b

For further questions requesting to ask on https://github.com/espressif/esp-aws-iot

Thanks

ssravindhar, 2021-09-02:
Hi @kolipakakondal 

Thank you so much for your replay .
I will check and update ASAP.

Thanks & Regards

Ravindhar S