espressif/idf-eclipse-plugin #354: Hi friends #353

ssravindhar, 2021-09-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

ssravindhar, 2021-09-03:
[image]

ssravindhar, 2021-09-03:
[image]
If i esp-idf i got this error

kolipakakondal, 2021-09-03:
Hi @ssravindhar Could you please create this issue on https://github.com/espressif/esp-aws-iot repo as it's more relevant there

ssravindhar, 2021-09-03:
OK @kolipakakondal 

I will create ....

ssravindhar, 2021-09-03:
Hi @kolipakakondal  I created this issue in https://github.com/espressif/esp-aws-iot  .

Thanks for your support.

closed 2021-09-03 by ssravindhar