espressif/idf-eclipse-plugin #355: 这个插件不支持arm64处理器进行开发？

LailuoA77, 2021-09-03:
root@raspberrypi:~# /usr/bin/python /home/pi/esp/esp-idf-v4.3/tools/idf_tools.py install
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
ERROR: tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64

LailuoA77, 2021-09-03:
[image]

kolipakakondal, 2021-09-06:
Hello @LailuoA77 

Your platform linux-arm64 is not supported. Kindly check this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#list-of-idf-tools

kolipakakondal, 2022-02-15:
Hello,

Please track https://github.com/espressif/esp-idf/issues/6432 to get the latest update on the aarch64 support. I am closing this issue as this is not specific to the eclipse plugin.

Thank you

closed 2022-02-15 by kolipakakondal