espressif/idf-eclipse-plugin #362: Missing requirement ESP-IDF Core on M1 mac

malja, 2021-09-21:
**Describe the bug**
When installing Eclipse IDF plugin, following [this readme](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) on M1 Mac, installation fails with following error:

    Cannot complete the install because one or more required items could not be found.
    Software being installed: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
    Missing requirement: ESP-IDF Core 1.0.1.202108090839 (com.espressif.idf.core 1.0.1.202108090839) requires 'osgi.bundle; org.eclipse.cdt.cmake.is.core 0.0.0' but it could not be found

    Cannot satisfy dependency:
        From: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
        To: org.eclipse.equinox.p2.iu; com.espressif.idf.core [1.0.1.202108090839,1.0.1.202108090839]

**To Reproduce**
Steps to reproduce the behavior:
1. Install lastest Eclipse CTD (2021-09)
2. Click Help -> Install new software
3. Add URL of the stable realease: https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
4. Select all packages available in IDF repository.
5. Click Install.

**Expected behavior**
I would expect that the IDF plugin would be installed.

**Environment Information:** 
https://gist.github.com/malja/5999f4a8fcbf9165a0f37059f7a0d0af

vchatchai, 2021-09-21:
I have the same problem on Ubuntu 20.04.2 LTS.

embeddedforce, 2021-09-21:
Same problem on Windows 10.

kolipakakondal, 2021-09-22:
Hello All,

Thanks for reporting this. Eclipse 2021-09 version is not yet supported by the IEP plugin due to some breaking changes at the CDT level. Kindly use Eclipse 2021-06 version.

Will keep you posted on this

Thanks

kolipakakondal, 2021-10-06:
Duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/363

closed 2021-10-06 by kolipakakondal