espressif/idf-eclipse-plugin #363: Cannot install idf-eclipse-plugin on 2021-09 version of Eclipse IDE

embeddedforce, 2021-09-21:
When I'm trying to install idf-eclipse-plugin, on clean 2021-09 version of Eclipse IDE, I'm getting the following error:

> Cannot complete the install because one or more required items could not be found.
>   Software being installed: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
>   Missing requirement: ESP-IDF Core 1.0.1.202108090839 (com.espressif.idf.core 1.0.1.202108090839) requires 'osgi.bundle; org.eclipse.cdt.cmake.is.core 0.0.0' but it could not be found
>   Cannot satisfy dependency:
>     From: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
>     To: org.eclipse.equinox.p2.iu; com.espressif.idf.core [1.0.1.202108090839,1.0.1.202108090839]

Easy way to reproduce:
1. Make clean Install of Eclipse IDE for C\C++ Developers 2021-09 (4.21.0)
1.1 (Optional) Update to the latest versions of plugins (Help->Check for Updates). Actualy it doesn't affect the problem...
2. Open "Install new software dialog" (Help ->Install new software...)
3. Check to install "Espressif IDF Plugin for Eclipse" from "https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/"
4. Boom! You get it!

kolipakakondal, 2021-09-22:
Hi @embeddedforce 

Thanks for reporting this. Eclipse 2021-09 version is not yet supported by the IEP plugin due to some breaking changes at the CDT level. Kindly use Eclipse 2021-06 version.

Thanks

sigmaaa, 2021-10-12:
Hi @andylemay,

Sorry for the late reply. What version of Eclipse CDT do you have? You are probably having the same problem because your CDT plugin is updated even on your eclipse 2020-09. You can check this by going to `Help -> About Eclipse IDE -> clicking the Eclipse CDT icon`

andylemay, 2021-10-12:
Hi @sigmaaa,
My earlier eclipse version is 2020-06.  Here are the CDT version details.
C/C++ Development Tools
Version: 9.11.1.202006011430
Build id: 20200601-1605

kolipakakondal, 2021-10-12:
Hi @andylemay It will be available sometime this week or early next week.

kolipakakondal, 2021-11-10:
We have released v2.3.0 and that would support Eclipse 2021-09. Please get the latest changes using the update site.
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2021-11-10 by kolipakakondal

Maksons, 2021-11-10:
Got another related problem:
[image]

kolipakakondal, 2021-11-12:
Hi @Maksons 

What is the eclipse version you're using? Could you please make sure you're having Eclipse 2021-06 and higher

Maksons, 2021-11-12:
@kolipakakondal  
I had Eclipse 2021-06 and check for updates and update everything except plugins fail to update (reported earlier).
After update I see Eslipse for C/C++ 2021-09 and  Eclipse IDE for Espressif IoT Development Framework(IDF)  2.3.0.202111091633 .
And that work for me ( compiling code / etc).

And you can see : 
`Embedded C/C++ Cross Compiler Common (incubation) 1.3.3.202010292017 `
can not be updated ( installed ) because of dependency 

    From: Espressif IDF Plugins for Eclipse 2.3.0.202111091633 (com.espressif.idf.feature.feature.group 2.3.0.202111091633)
    To: org.eclipse.equinox.p2.iu; ilg.gnumcueclipse.managedbuild.cross [1.3.2.202007271621,1.3.2.202007271621]

If I understand correct, correct me if I'm wrong.
Question - is there incompatibilities between 
ESP-IDF Plugins for Eclipse 2.3.0.202111091633 
and new
Embedded C/C++ Cross Compiler Common (incubation) 1.3.3.202010292017
?

Another thing I can imagine for side plugin update fail is that I have also another "Eclipse for **Embedded** C/C++" which could had installed `Embedded C/C++ Cross Compiler Common` and try to keep it up to date.
But those 2 Eclipse installation should not be related, but I'm not 100% sure .

kolipakakondal, 2021-11-12:
> If I understand correct, correct me if I'm wrong.
> Question - is there incompatibilities between
> ESP-IDF Plugins for Eclipse 2.3.0.202111091633
> and new
> Embedded C/C++ Cross Compiler Common (incubation) 1.3.3.202010292017
> ?

No, by default IEP plugin installs the 5.1.1. I was trying to install the latest one also `https://download.eclipse.org/embed-cdt/releases/6.1.2/p2` and this seems to be working without any issue

I just did this experiment:

1. Downloaded 2021-09 CPP
2. Installed IEP 2.3.0 along with other plugins suggested as part of the update site
3. Then I was trying to install the latest embedded-cdt plugins using `https://download.eclipse.org/embed-cdt/releases/6.1.2/p2` and I have no issue in installing them

Maybe I am missing something.

Marcarleto, 2022-07-15:
Hi kolipakakondal,

I'm using MCUXpresso IDE v11.5.1 wich I belive is running Eclipse version 2021-06 that should work, right? But I'm having similar problems here:

Cannot complete the install because one or more required items could not be found.
  Software being installed: Espressif IDF Plugins for Eclipse 2.5.0.202206080313 (com.espressif.idf.feature.feature.group 2.5.0.202206080313)
  Missing requirement: ESP-IDF Core 1.0.1.202206080313 (com.espressif.idf.core 1.0.1.202206080313) requires 'osgi.bundle; org.eclipse.cdt.cmake.core 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.5.0.202206080313 (com.espressif.idf.feature.feature.group 2.5.0.202206080313)
    To: org.eclipse.equinox.p2.iu; com.espressif.idf.core [1.0.1.202206080313,1.0.1.202206080313]

[image]

Samg381, 2022-09-22:
This is still an issue.