espressif/idf-eclipse-plugin #366: Invalid Certificate

PabloMMS, 2021-10-01:
[1/313] Performing build step for 'bootloader'
ninja: no work to do.
[2/311] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D C:\Users\pablo\Eclipse\eclipse-workspace\Lab_SyM_Blink\build\esp-idf\mbedtls && python C:/msys2/home/pablo/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/msys2/home/pablo/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in C:/msys2/home/pablo/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Executing action: flash
Running ninja in directory c:\users\pablo\eclipse\eclipse-workspace\lab_sym_blink\build
Executing "ninja flash"...

Anybody knows how to solve this, thanks btw

kolipakakondal, 2021-10-01:
Hi @PabloMMS 

This could help you https://github.com/espressif/esp-idf/issues/7621

kolipakakondal, 2021-10-01:
1. Go to project
2. Open `sdkconfig` editor
3. Navigate to `Component Config` > `mbedLTS` > `Certificate Bundle` > Uncheck `Enable trusted root certificate bundle` option
4. Rebuild the project

Please find the screenshot to disable the root certificate bundle.

[image]

PabloMMS, 2021-10-01:
Thanks, really helped me out
Didn't expect to use the sdk conf to solve it

closed 2021-10-01 by PabloMMS

itwax, 2021-10-06:
Thanks.  But what are we losing by disabling Trusted Root Certificate ? 
It is there for some security reason. 
Espressif should also provide an alternative.

TobiasBoeschMiele, 2021-10-06:
> Thanks. But what are we losing by disabling Trusted Root Certificate ? It is there for some security reason. Espressif should also provide an alternative.

I also ask myself if this is really a solution to the problem or more a workaround.