espressif/idf-eclipse-plugin #370: Plugin Install Failure (Missing requirement/Cannot satisfy dependency)

BMaj7, 2021-10-05:
**Describe the bug**
Following the description to install plugin under [https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url](doc) continues to fail with error message

_Cannot complete the install because one or more required items could not be found.
  Software being installed: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
  Missing requirement: ESP-IDF Core 1.0.1.202108090839 (com.espressif.idf.core 1.0.1.202108090839) requires 'osgi.bundle; org.eclipse.cdt.cmake.is.core 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.2.0.202108131352 (com.espressif.idf.feature.feature.group 2.2.0.202108131352)
    To: org.eclipse.equinox.p2.iu; com.espressif.idf.core [1.0.1.202108090839,1.0.1.202108090839]_

**To Reproduce**
Exactly as [https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url](described)
Happens also when trying to install via Marketplace

**Expected behavior**
Installation of plugin as described

**Environment Information:** 
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2021-09 (4.21.0)
Build id: 20210910-1417

[config.txt](https://github.com/espressif/idf-eclipse-plugin/files/7284860/config.txt)

**ESP-IDF Tools Information:**
Failed to install

**IDF Eclipse Plugin Version:** 
Failed to install

**Eclipse Error log:** 
[log.txt](https://github.com/espressif/idf-eclipse-plugin/files/7284797/log.txt)

BMaj7, 2021-10-05:
Duplicate of [https://github.com/espressif/idf-eclipse-plugin/issues/363](363)

kolipakakondal, 2021-10-06:
Eclipse 2021-09 is not yet supported. Please see my post here https://github.com/espressif/idf-eclipse-plugin/issues/363

Closing as it's a duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/363