espressif/idf-eclipse-plugin #408: changing com port in launch configuration has no effect on run/debug port used

ralphde, 2021-11-23:
changing com port in launch configuration has no effect on run/debug port used.
In my case com12 seem hard coded

`[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\ESP-IDF-4_3\esp-idf-v4.3.1\components\esptool_py && C:\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/ESP-IDF-4_3/esp-idf-v4.3.1" -D SERIAL_TOOL="python C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py --chip esp32s2" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/ralph/eclipse-workspace/restful_server/build" -P C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32s2 -p COM12 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 restful_server.bin
esptool.py v3.1-dev
Serial port COM12
Traceback (most recent call last):
  File "C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4535, in <module>
    _main()
  File "C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4528, in _main
    main()
  File "C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 4027, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 120, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py", line 310, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM12': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
CMake Error at run_serial_tool.cmake:50 (message):
  python
  C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py
  --chip esp32s2 failed

FAILED: CMakeFiles/flash 
cmd.exe /C "cd /D C:\ESP-IDF-4_3\esp-idf-v4.3.1\components\esptool_py && C:\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/ESP-IDF-4_3/esp-idf-v4.3.1" -D SERIAL_TOOL="python C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/esptool/esptool.py --chip esp32s2" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/ralph/eclipse-workspace/restful_server/build" -P C:/ESP-IDF-4_3/esp-idf-v4.3.1/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Executing action: flash
Running ninja in directory c:\users\ralph\eclipse-workspace\restful_server\build
Executing "ninja flash"...
`
WIN10pro
Eclipse 2021-06 from win installer
Both IDF 2.2 and 2.3 have the same issue

This is a fatal issue what do I have to do to fix this

ralphde, 2021-11-26:
Looks like the issue was provoked by creating a launch target without a name.

the pull down menu looks then like this:
[image]

after reinstall the whole tool chain the problem was fixed

kolipakakondal, 2022-02-15:
Hello, This issue got fixed with v2.4.0. Please get the latest changes using the update site and see that helps. Thanks

closed 2022-02-15 by kolipakakondal