espressif/idf-eclipse-plugin #412: Paths for Eclipse on Ubuntu 20.4

rdkluijv, 2021-12-01:
A fresh minimal install of Ubuntu 20.4. Installed all the needed packages and downloaded the espressif software. Build the Hello World application, downloaded it and checked with the monitor. Downloaded the Eclipse installer and installed Eclipse.
Installed the plugin according to the manual. Did a manual install using the existing directory. Restarted the computer.
Tried to use Tools manager but that did not work. Even the Browse button did not respond. Started configuring the environment variables. Changed it to the "new" paths for the espressif software. (At the end I pasted a copy.) There is some trouble here because cmake is in a different location and there is no ninja. 
Created a new Espressif IDF project. No buttons Next and Back so I could not load a Template. Configured the project and hit the hammer. This resulted in: "Build not configured correctly". Could change that to: 
"Building in: /home/rdkluijv/eclipse-workspace/Test00/build
No toolchain file found"
After I had added Core Build Toolchains. With the next step "Configuring CMake Toolchain" and restarting Eclipse I get the first error message back. Now I created a new project. Building this new project resulted in this error:
"Building in: /home/rdkluijv/eclipse-workspace/Test01/build
Configuring in: /home/rdkluijv/eclipse-workspace/Test01/build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=/home/rdkluijv/esp/esp-idf/tools/cmake/toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 /home/rdkluijv/eclipse-workspace/Test01
Failure running cmake: "
Here I get stuck.

$HOME/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:
/Users/user-name/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:
$HOME/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:
/Users/user-name/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:

/Users/user-name/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:
$OPENOCD/bin:
/Users/user-name/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:

/Users/user-name/.espressif/tools/ninja/1.9.0/:
$IDF_PYTHON_ENV_PATH/bin:
/Users/user-name/.espressif/python_env/idf4.0_py3.7_env/bin:
$IDF_PATH/tools:
/Users/user-name/esp/esp-idf/tools:
$PATH

kolipakakondal, 2021-12-22:
> Got it working. $PATH does not get expanded so /usr/bin had to be added to the path.

Sorry for replying late on this. Is that something you have added manually to the eclipse CDT build environment variables or you have used Tools manager to install the tools, even then $PATH is not expanded variable value?

rdkluijv, 2021-12-30:
Installing problem was caused by the new release (5). I configured the environmental variable manual.
When I saw $PATH in the PATH below at the end I assumed I also could use $HOME as well.
So:
IDF_PATH
/Users/user-name/esp/esp-idf
Became:
$HOME/esp/esp-idf
That did not work.
To get other components to work I had to install a lower release and that worked perfect.

PATH
/Users/user-name/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:/Users/user-name/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/Users/user-name/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:/Users/user-name/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/Users/user-name/.espressif/tools/ninja/1.9.0/:/Users/user-name/.espressif/python_env/idf4.0_py3.7_env/bin:/Users/user-name/esp/esp-idf/tools:$PATH

alirana01, 2022-05-05:
Closing as unable to reproduce with latest release

closed 2022-05-05 by alirana01