espressif/idf-eclipse-plugin #414: icon becomes ugly under hidpi fractional scaling

erusyd, 2021-12-09:
[image]
System: linux mint 20.2

kolipakakondal, 2021-12-20:
Hi @alirana01  @sigmaaa Do you have a Linux box to verify this?

erusyd, 2021-12-20:
My fractional scaling configurations:
(1) Display -> Settings -> turn on "Enable fractional scaling controls(experimental)"
(2) Display -> Layout -> in "User interface scale", select "Double (Hi-DPI)" -> then turn on "Fractional scaling", and select 125%

kolipakakondal, 2021-12-22:
Thanks for the details @erusyd 

Looks like these problems are there at the eclipse level as mentioned here https://bugs.eclipse.org/bugs/show_bug.cgi?id=563923

As suggested in the bug, could try setting `-Dswt.autoScale=125` in the `eclipse.ini` file and restart your eclipse and see if that makes any difference

Please refer to the below resource to configure eclipse.ini file
https://wiki.eclipse.org/Eclipse.ini refer

sigmaaa, 2021-12-22:
Hi @erusyd,

I was able to reproduce your issue. In my case, setting `-Dswt.autoScale=100` helped to fix it. Looks like, with this option, scaling has no impact on eclipse icons

erusyd, 2021-12-22:
> Hi @erusyd,
> 
> I was able to reproduce your issue. In my case, setting `-Dswt.autoScale=100` helped to fix it. Looks like, with this option, scaling has no impact on eclipse icons

Verified @sigmaaa. Thank you guys!

closed 2021-12-22 by erusyd