espressif/idf-eclipse-plugin #423: ESP32-C3 Eclipse OpenOCD timed out on larger projects (OCD-458)

JeroenvdBE, 2021-12-23:
**Environment**

- **Development Kit**: esp32-c3-devkitc-02
- **Module**: ESP32-C3-WROOM-02
- **Debug Adapter**: USB JTAG Builtin
- **OpenOCD version**: v0.10.0-esp32-20210401 (default), v0.10.0-esp32-20211111 and v0.11.0-esp32-20211220, the results are the same.
- **Operating System**: Windows
- **IDE**: Eclipse (https://github.com/espressif/idf-installer : esp-idf-tools-setup-offline-2.12.exe)
- J**TAG Clock** Default
- **ESP-IDF version**: 4.3.1
- **Debug setup**: ESP-IDF GDB OpenGDB Debugging, using build in USB JTAG (esp32c3-builtin.cfg)

Tested with different OpenOCD version: v0.10.0-esp32-20210401 (default), v0.10.0-esp32-20211111 and v0.11.0-esp32-20211220, the results are the same.

**Problem Description**
Small examples like "hello_world" run fine (flashing & debugging)

Larger examples like the (WiFi) "scan" or (Bluetooth) "multi-adv" example give a OpenOCD timeout : " OpenOCD timed out". Flashing the application is then aborted it seems and debugging ofcourse will not work.

When starting penOCD manually, using the same parameters as Eclipse does, and waiting for OpenOCD to finish programming works fine. When we change the eclipse debug configuration and disable the "Start OpenOCD" locally and thus connect to the manually started OpenOCD the debugging of larger projects seem to work fine.

**Debug Logs**

[logoutput.txt](https://github.com/espressif/openocd-esp32/files/7763478/logoutput.txt)
Eclipse log (.log in workspace)

!ENTRY com.espressif.idf.debug.gdbjtag.openocd 4 1 2021-12-22 15:21:44.978
!MESSAGE C:\espidf\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/openocd.exe -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\espidf\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/share/openocd/scripts -f board/esp32c3-builtin.cfg -d3 -l openocd_log.txt -c program_esp_bins C:/Projects/EspressifWorkspace/multi-adv/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"

!ENTRY org.eclipse.cdt.dsf 4 10005 2021-12-22 15:21:59.982
!MESSAGE Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.concurrent.RequestMonitorWithProgress@7a042f53): Status ERROR: ilg.gnumcueclipse.debug.gdbjtag code=5010 Starting OpenOCD timed out. null' resulted in an error.
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2021-12-22 15:21:59.982
!MESSAGE Starting OpenOCD timed out.

!ENTRY org.eclipse.cdt.dsf.gdb 4 5012 2021-12-22 15:21:59.985
!MESSAGE Error in services launch sequence
!STACK 1
org.eclipse.core.runtime.CoreException: Starting OpenOCD timed out.
at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
at java.base/java.lang.Thread.run(Thread.java:831)
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2021-12-22 15:21:59.985
!MESSAGE Starting OpenOCD timed out.

kolipakakondal, 2022-02-14:
Hello, We have increased the default timeout for the OpenOCD server from 15 sec to 25 sec with the v2.4.1 release and also provided an option to users to configure the timeout using the preference page.

> Eclipse > Prefernces > Espressif > GDB server launch timeout

Please get the latest changes using the update site and see if that helps.

Thank you.

closed 2022-02-14 by kolipakakondal

TobiasBoeschMiele, 2022-02-17:
Thanks for reporting this and thank for providing a fix.
I have the same issue and an interesting observation with Plugin version 2.4.1:

When flashing via JTAG the issue with OpenOCD timeout occurs when I start the launch configuration. When I change the GDB timeout from 25 to 100 in the preferences the issue is gone.
Do I change this setting back to 25 the issue stays gone.

I expect it to be back when reverting the settings that made it go away.