espressif/idf-eclipse-plugin #425: use of ESP32-DevKitM-1U with JTAG esp-prog

gregrasche, 2021-12-28:
I am trying to single step the debugger in hello_world using esp-prog on ESP32-DevKitM-1U which is an esp32s2 chipset.  I have wired the 4 JTAG pins and the GND between the boards, where I have a laptop connected to both USB ports on the two boards.  I am using the esp-idf-v4.3.2 with eclipse 2109.  

I can program hello_world project in the DevKit as needed, and see that I can flash the project properly, as "hello world" is seen at the console.  

What JTAG device is expected in 'lsusb' response?

**To Reproduce**
I am following this to set up the debug session
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse 

**Expected behavior**
JTAG debug session with ability to single step in main.c

**ESP-IDF Tools Information:**
esp-idf-v4.3.2

**IDF Eclipse Plugin Version:** 
2109 with Espressif plugin

**Eclipse Error log:** 
Cannot connect to JTAG debug session.  Get TCP/IP error.  I am not sure what TCP/IP has to do with it.

gregrasche, 2021-12-28:
Here is the error log

Error in final launch sequence:

Failed to execute MI command:
-target-select remote localhost: 3333

Error message from debugger back end:
localhost: 3333: Connection timed out.
Failed to execute MI command:
-target-select remote localhost: 3333

Error message from debugger back end:
localhost: 3333: Connection timed out.
localhost: 3333: Connection timed out.

gregrasche, 2021-12-29:
Is there a CFG similar to esp32s2-kaluga-1.cfg for the  ESP32-DevKitM-1U which is an esp32s2 chipset ?

kolipakakondal, 2021-12-29:
Hello @gregrasche 

Please check the specific configuration file for the board here https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/jtag-debugging/tips-and-quirks.html#configuration-of-openocd-for-specific-target

Can you try with `interface/ftdi/esp32_devkitj_v1.cfg` this is a JTAG adapter configuration file for ESP-Prog boards.

> To Reproduce
> I am following this to set up the debug session
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse

Could you use the `ESP-IDF GDB OpenOCD Debugging` option? Please refer this doc https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md

Also, provide us the OpenOCD debug log to check further
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-generate-openocd-log-

gregrasche, 2021-12-30:
Thank you for the CFG.  I will see if interface/ftdi/esp32_devkitj_v1.cfg works for ESP-DevKitM.  The ttyUSB0 is created by the ESP-DevKitM and works as desired.

I have wired the 4 JTAG pins and the GND between the boards, where I have a laptop connected to the USB ports on the two boards. The lsusb shows that only ttyUSB1and ttyUSB2 are added when the ESP-PROG is plugged into the USB, where both tty will not be used. 

I am going through the OpenOcd setup you mentioned, but want to get the HW right first.  Is this a functional hardware setup of the two boards?

gregrasche, 2021-12-30:
Here is my wiring cable

GND               DevKit J3-1         ESP-PROG J2-9
TMS               DevKit J3-7         ESP-PROG J2-2
TDI                 DevKit J3-8         ESP-PROG J2-8
TDO               DevKit J3-9         ESP-PROG J2-6
TCK               DevKit J3-10       ESP-PROG J2-4

gregrasche, 2021-12-30:
I just did a new tool setup with ESP-IDF 4.3.2 version, still on Eclipse 2109. 

I wired it all up as I stated above, but get the following in the OCD log last lines.  

271 Debug: 307 142 command.c:355 register_command_handler(): registering 'test_image'...
272 Debug: 308 142 command.c:355 register_command_handler(): registering 'reset_nag'...
273 Debug: 309 142 command.c:355 register_command_handler(): registering 'ps'...
274 Debug: 310 142 command.c:355 register_command_handler(): registering 'test_mem_access'...
275 Debug: 311 142 ftdi.c:728 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
276 Error: 312 151 mpsse.c:190 open_matching_device(): libusb_open() failed with LIBUSB_ERROR_ACCESS
277 Error: 313 151 mpsse.c:216 open_matching_device(): no device found
278 Error: 314 151 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
279 Error: 315 158 mpsse.c:216 open_matching_device(): no device found
280 Error: 316 158 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
281 Debug: 317 158 command.c:630 run_command(): Command 'init' failed with error code -100
282 Debug: 318 158 command.c:143 script_debug(): command - echo echo ** OpenOCD init failed **
283 User : 320 158 command.c:770 jim_echo(): ** OpenOCD init failed **
284 Debug: 321 158 command.c:143 script_debug(): command - shutdown shutdown error
285 User : 323 158 server.c:742 handle_shutdown_command(): shutdown command invoked
286 Debug: 324 158 command.c:630 run_command(): Command 'shutdown' failed with error code -4
287 Debug: 325 158 command.c:143 script_debug(): command - echo echo ** Flashing Failed **
288 User : 327 158 command.c:770 jim_echo(): ** Flashing Failed **
289 User : 328 158 options.c:57 configuration_output_handler(): -1User : 329 158 options.c:57 configuration_output_handler():
290 Debug: 330 158 command.c:143 script_debug(): command - echo echo Started by GNU MCU Eclipse
291 User : 332 158 command.c:770 jim_echo(): Started by GNU MCU Eclipse
292 Info : 333 158 server.c:311 add_service(): Listening on port 6666 for tcl connections
293 Info : 334 159 server.c:311 add_service(): Listening on port 4444 for telnet connections
294 Debug: 335 159 command.c:143 script_debug(): command - init init
295 Debug: 337 159 esp_xtensa.c:194 esp_xtensa_target_deinit(): start
296 Debug: 338 159 xtensa.c:2533 xtensa_target_deinit(): start
297 Debug: 339 159 server.c:611 sig_handler(): Ignored extra Signal 6 

The ttyUSB0 (DevKitM), ttyUSB1 (ESP-PROG), and ttyUSB2 (ESP-PROG) exist and are all writable.  

How do I tell the eclipse that it should use ttyUSB0 DTR & RTS to control the RST and BOOT signals on the board?  Is that the problem?  I do not understand  why I am getting
              command.c:770 jim_echo(): ** OpenOCD init failed **

gregrasche, 2022-01-02:
There are other issues but this fixed the JTAG open issue

https://docs.platformio.org/en/latest/faq.html#faq-udev-rules  

99-platformio-udev.rules¶
Linux users have to install udev rules for PlatformIO supported boards/devices

gregrasche, 2022-01-02:
Here is my log where the JTAG debug session opened fine, but is shut down because of signal 15 (SIGTERM).  Any ideas on how to debug this?

Debug: 4717 5781 esp32s2.c:317 esp32s2_deassert_reset(): esp32s2: begin
Debug: 4718 5781 xtensa.c:864 xtensa_deassert_reset(): esp32s2 halt=0
Debug: 4719 5781 xtensa.c:726 xtensa_smpbreak_write(): esp32s2: write smpbreak set=0x0 clear=0x630000
Debug: 4720 5782 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event reset-deassert-post
Debug: 4721 5782 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event reset-end
Info : 4722 5782 xtensa.c:1810 xtensa_poll(): esp32s2: Core was reset.
Debug: 4723 5782 esp_xtensa.c:215 esp_xtensa_poll(): esp32s2: Clear debug stubs info 
User : 4724 5782 options.c:57 configuration_output_handler(): 0User : 4725 5782 options.c:57 configuration_output_handler(): 
Debug: 4726 5782 command.c:143 script_debug(): command - echo echo Started by GNU MCU Eclipse
User : 4728 5783 command.c:770 jim_echo(): Started by GNU MCU Eclipse
Info : 4729 5783 server.c:311 add_service(): Listening on port 6666 for tcl connections
Info : 4730 5783 server.c:311 add_service(): Listening on port 4444 for telnet connections
Debug: 4731 5783 command.c:143 script_debug(): command - init init 
Debug: 4733 13673 server.c:609 sig_handler(): Terminating on Signal 15
Debug: 4734 13673 command.c:143 script_debug(): command - shutdown shutdown
User : 4736 13673 server.c:742 handle_shutdown_command(): shutdown command invoked
Debug: 4737 13674 esp_xtensa.c:194 esp_xtensa_target_deinit(): start
Debug: 4738 13674 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 4739 13674 target.c:2012 target_free_all_working_areas_restore(): freeing all working areas
Debug: 4740 13674 target.c:1753 print_wa_layout():    0x40030000-0x40033fff (16384 bytes)

gregrasche, 2022-01-02:
I had this for JTAG Config Options
 -s ${openocd_path}/share/openocd/scripts -c 'set ESP32_FLASH_VOLTAGE 3.3' -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32s2.cfg -d3 -l oocd.log

but the debugger terminated as above with a Signal 15.  Removal of the "-d3 -l oocd.log" made the JTAG able to single step!

gregrasche, 2022-01-02:
JTAG single step works using Eclipse works for DevKitM with esp-idf-v4.3.2 on an esp32s2, using esp-prog for JTAG.

closed 2022-01-02 by gregrasche

gerekon, 2022-01-10:
> There are other issues but this fixed the JTAG open issue
> 
> https://docs.platformio.org/en/latest/faq.html#faq-udev-rules
> 
> 99-platformio-udev.rules¶ Linux users have to install udev rules for PlatformIO supported boards/devices

@kolipakakondal When setting up tools on Linux we need to copy `60-openocd.rules` as it is suggested [here](https://github.com/espressif/openocd-esp32/blob/master/contrib/60-openocd.rules#L1).

As far as I know VsCode does this during onboarding process.