espressif/idf-eclipse-plugin #427: Open OCD timed out when spiffs image is used

DigKleppe, 2021-12-29:
I have a small project containing an spiffs image. To start I issue a  "idf.py build flash" command to flash all needed images, including the spiffs one. App works ok.

JTAG Debugging with ESP-IDF- GDB openOCD configuration with default settings fails.
I get this message on the serial port:
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6724
load:0x40078000,len:14812
load:0x40080400,len:3792
entry 0x40080694
<0x1b>[0;32mI (52) boot: ESP-IDF v4.4-dev-3675-g35b20cadce 2nd stage bootloader<0x1b>[0m
<0x1b>[0;32mI (53) boot: compile time 19:16:12<0x1b>[0m
<0x1b>[0;32mI (53) boot: chip revision: 3<0x1b>[0m
<0x1b>[0;32mI (58) boot_comm: chip revision: 3, min. bootloader chip revision: 0<0x1b>[0m
<0x1b>[0;32mI (65) boot.esp32: SPI Speed      : 40MHz<0x1b>[0m
<0x1b>[0;32mI (69) boot.esp32: SPI Mode       : DIO<0x1b>[0m
<0x1b>[0;32mI (74) boot.esp32: SPI Flash Size : 4MB<0x1b>[0m
<0x1b>[0;32mI (78) boot: Enabling RNG early entropy source...<0x1b>[0m
<0x1b>[0;32mI (84) boot: Partition Table:<0x1b>[0m
<0x1b>[0;32mI (87) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m
<0x1b>[0;32mI (95) boot:  0 nvs              WiFi data        01 02 00009000 00006000<0x1b>[0m
<0x1b>[0;32mI (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000<0x1b>[0m
<0x1b>[0;32mI (110) boot:  2 factory          factory app      00 00 00010000 00300000<0x1b>[0m
<0x1b>[0;32mI (117) boot:  3 storage          Unknown data     01 82 00310000 000d0000<0x1b>[0m
<0x1b>[0;32mI (125) boot: End of partition table<0x1b>[0m
<0x1b>[0;32mI (129) boot_comm: chip revision: 3, min. application chip revision: 0<0x1b>[0m
<0x1b>[0;32mI (136) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=187fch (100348) map<0x1b>[0m
<0x1b>[0;32mI (181) esp_image: segment 1: paddr=00028824 vaddr=3ffb0000 size=0380ch ( 14348) load<0x1b>[0m
<0x1b>[0;32mI (187) esp_image: segment 2: paddr=0002c038 vaddr=40080000 size=03fe0h ( 16352) load<0x1b>[0m
<0x1b>[0;32mI (194) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=8a784h (567172) map<0x1b>[0m
<0x1b>[0;31mE (401) esp_image: invalid segment length 0xffffffff<0x1b>[0m
<0x1b>[0;31mE (401) boot: Factory app partition is not bootable<0x1b>[0m
<0x1b>[0;31mE (401) boot: No bootable app partitions in the partition table<0x1b>[0m
ets Jul 29 2019 12:21:46
```

It seems something is overwritten? Which files are flashed?

Debugging with a old style ESP-ESP32- openOCD  configuration works fine.
In this configuration i program the esp with openocd -c "program_esp .../test2812/build/app.bin 
0x10000 verify" config option. So only the app partition is written.

Other projects without the spiffs partition are working OK

Eclipse log:
```

!ENTRY ilg.gnumcueclipse.managedbuild.cross 4 1 2021-12-29 19:33:22.338
!MESSAGE /home/dig/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/bin/openocd -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s /home/dig/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/bin/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c program_esp_bins /mnt/HDlinux/projecten/esp32/test2812/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"

!ENTRY org.eclipse.cdt.dsf 4 10005 2021-12-29 19:33:38.336
!MESSAGE Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.concurrent.RequestMonitorWithProgress@60d07734): Status ERROR: ilg.gnumcueclipse.debug.gdbjtag code=5010 Starting OpenOCD timed out. null' resulted in an error.
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2021-12-29 19:33:38.336
!MESSAGE Starting OpenOCD timed out.

!ENTRY org.eclipse.cdt.dsf.gdb 4 5012 2021-12-29 19:33:38.341
!MESSAGE Error in services launch sequence
!STACK 1
org.eclipse.core.runtime.CoreException: Starting OpenOCD timed out.
	at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
	at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:831)
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2021-12-29 19:33:38.341
!MESSAGE Starting OpenOCD timed out.
```
[template291221.zip](https://github.com/espressif/idf-eclipse-plugin/files/7790252/template291221.zip)

kolipakakondal, 2022-02-14:
Hello, We have increased the default launch timeout for the OpenOCD server from 15 sec to 25 sec with the v2.4.1 release and also provided an option to the users to configure the timeout using the preference page.

> Eclipse > Prefernces > Espressif > GDB server launch timeout

Please get the latest changes using the update site and see if that helps.

Thank you.

closed 2022-02-14 by kolipakakondal

DigKleppe, 2022-02-14:
Hi,

See [https://github.com/espressif/idf-eclipse-plugin/issues/480](url)