espressif/idf-eclipse-plugin #431: ESP-IDF: Application Size Analyis Crashes with  Cannot invoke "java.lang.Long.longValue()" because the return value of "org.json.simple.JSONObject.get(Object)" is null

expresspotato, 2022-01-06:
**Describe the bug**
Right click on a relevant project, go to ESP-IDF: Application Size Analysis, results in Null Pointer Exception with ESP-IDF 4.3.2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: An open Eclipse project, with esp-idf nature
2. Right click on: ESP-IDF: Application size analysis
3. See error

**Expected behavior**
Expected to see the chart.

**Environment Information, Tools, IDF and Eclipse IDF Version ** 
Executing /home/kevin/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/kevin/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20211111 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

IDF_PATH: /home/kevin/esp-idf
IDF_PYTHON_ENV_PATH: /home/kevin/.espressif/python_env/idf4.3_py3.8_env
PATH: /home/kevin/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin:/home/kevin/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/kevin/.espressif/tools/cmake/3.13.4/bin:/home/kevin/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/home/kevin/.espressif/tools/ninja/1.9.0/:/home/kevin/esp-idf-v4.3.2/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/home/kevin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/home/kevin/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/home/kevin/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/home/kevin/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/kevin/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/kevin/.espressif/tools/cmake/3.16.4/bin:/home/kevin/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin:/home/kevin/.espressif/tools/ninja/1.10.2/

Operating System: linux
Java Runtime Version: 17.0.1+12
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.3.0.202110080828
ESP-IDF v4.3.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.10
Python declared in PATH variable: Python 3.8.10

**Eclipse Error log:** 
!ENTRY com.espressif.idf.core 1 0 2022-01-06 18:40:45.703
!MESSAGE Mapping file path /home/kevin/workspace/liberty/build/liberty.map

!ENTRY com.espressif.idf.core 1 0 2022-01-06 18:40:45.774
!MESSAGE Editor input:/home/kevin/workspace/liberty/build/liberty.map

!ENTRY com.espressif.idf.core 1 0 2022-01-06 18:40:45.804
!MESSAGE [/home/kevin/.espressif/python_env/idf4.3_py3.8_env/bin/python, /home/kevin/esp-idf/tools/idf_size.py, /home/kevin/workspace/liberty/build/liberty.map, --json]

!ENTRY com.espressif.idf.core 1 0 2022-01-06 18:40:46.024
!MESSAGE {
    "dram_data": 15800,
    "dram_bss": 6512,
    "dram_rodata": 0,
    "dram_other": 0,
    "used_dram": 22312,
    "dram_total": 124580,
    "used_dram_ratio": 0.17909776850216727,
    "dram_remain": 102268,
    "iram_vectors": 1027,
    "iram_text": 100352,
    "iram_other": 0,
    "used_iram": 101379,
[... 88 lines cut ...]
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4584)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1154)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1045)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1440)

I should note idf.py size-components works fine.

sigmaaa, 2022-01-07:
Hi @expresspotato,
the fix for this issue is available on the master branch. We are planning to make a new release soon, however, you can try to install the master branch build:
https://github.com/espressif/idf-eclipse-plugin/suites/4814845567/artifacts/135598166
To install, follow the instructions [here](https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile).

Also, you can try this build with espressif-ide for linux:
https://github.com/espressif/idf-eclipse-plugin/suites/4814845567/artifacts/135598167

ApplicationNinja, 2022-01-22:
FYI, I downloaded, compiled and installed the master branch an hour ago, and this bug still exists.

Operating System: mac os x
Java Runtime Version: 16.0.2+7-67
Eclipse Version: 4.21.0.v20210906-0500
Eclipse CDT Version: 10.4.0.202106130052
IDF Eclipse Plugin Version: 2.4.0.202201170830
ESP-IDF v4.3.2
Python set for IDF_PYTHON_ENV: Python 3.8.9
Python declared in PATH variable: Python 2.7.18

sigmaaa, 2022-01-23:
Hi @ApplicationNinja,

thank you for verifying and for the information. My bad, I was testing it on release/4.3 instead of 4.3.2. I was able to reproduce this problem on 4.3.2 and prepared a fix for it in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5028616382/artifacts/148260397). Please, let me know if this works for you. Thanks

ApplicationNinja, 2022-01-23:
> Hi @ApplicationNinja,
> 
> thank you for verifying and for the information. My bad, I was testing it on release/4.3 instead of 4.3.2. I was able to reproduce this problem on 4.3.2 and prepared a fix for it in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5028616382/artifacts/148260397). Please, let me know if this works for you. Thank you

This is fixed in this build, thank you.

kolipakakondal, 2022-02-14:
Hello All, Fix is available as part of the IEP 2.4.0 release. Please get the latest changes using the update site to verify. Thank you.

closed 2022-02-14 by kolipakakondal