espressif/idf-eclipse-plugin #436: v2.3.0 Launch configuration can not saved

gaupen1186, 2022-01-15:
**Describe the bug**
Launch configuration can not saved.

**To Reproduce**
Steps to reproduce the behavior:
1, click the icon;
2, enable flash over jtag;
3, change arguments;
4, click ok to save.
5, click the icon again to check;
6, Not effective.

**Expected behavior**
can not download fw with builtin usb jtag.

**Screenshots**
[image]

**Environment Information:** 
[log.log](https://github.com/espressif/idf-eclipse-plugin/files/7874637/log.log)

**ESP-IDF Tools Information:**
Executing D:\ESP32\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\ESP32\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
release v2.3.0

windows 10 x64
eclipse c++ 2021.09
eps-idf v4.4 beta 1, Crosstool and openocd have been updated manually to use esp32-s3
command line download  & debug with builtin usb jtag test ok.
eclipse builtin usb jtag debug ok.
HW: ESP32-S3 -DevKitC-1 v1.0, builtin usb jtag

sigmaaa, 2022-01-19:
Hi @gaupen1186,

thank you for reporting the issue. Fix for it is available in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/4990322027/artifacts/146019743). To install, follow the instructions [here](https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url).

gaupen1186, 2022-01-20:
Great! I have test it and it's OK now!

closed 2022-01-20 by gaupen1186