espressif/idf-eclipse-plugin #440: Need to know what hardware changes are required to JTAG debug on an esp32-c3-DevKitM

gregrasche, 2022-01-17:
I read esp32-c3-DevKitM schematic, and see that USB can go to either the one of these
1) ESP32-C3 
2) FTDI chip

depending on the NC resistor installation.  NC are DNIs (Do Not Install) components on the esp32-c3-DevKitM.   The esp32-c3 has an internal JTAG device, and a serial port that I want to use for software debug.

Do we have to do this to the DevKit (disable FTDI, enable ESP32-C3 USB) to JTAG debug?
* remove R1/R4/R21/R22
* add R2/R3

We could need do this to
* Perform ESP32-C3 JTAG software debug over ESP32-C3 USB

Get ‘program interface’ access to ESP32-C3 UART U0 we might need to get a serial port TTL-to-USB voltage converter for J3-2/J3-3 for UART U0 of the esp32-s3 DevKitM

sigmaaa, 2022-01-18:
Hi @gregrasche,

> Do we have to do this to the DevKit (disable FTDI, enable ESP32-C3 USB) to JTAG debug?

The easiest way to start debugging JTAG is with a USB cable connected to the D+/D- USB pins of the ESP32-C3. 
In this case, you need to connect the USB to the following pins on the board:

- D+ - 19
- D- -18
- V_BUS - 5V
- GND - GND

If you want to use an internal USB JTAG interface, replacing R1/R4 with R2/R3 should be enough.

gregrasche, 2022-01-19:
If I do what was suggested 'replacing R1/R4 with R2/R3' do we need to get a serial port TTL-to-USB voltage converter to use ‘program interface’ to access to ESP32-C3 UART U0?
We might need to get for J3-2/J3-3 for UART U0 of the esp32-s3 DevKitM

gregrasche, 2022-01-20:
The esp32-c3_technical_reference_manual_en_v05.pdf section 26.2 says
       ROM startup code of the SoC contains code allowing the user to reprogram attached flash memory using this interface.

Does this mean the the USB UART that is presented by the esp32c3 can be used as the 'programming interface' using the command below?

    Idf.py -p /dev/ttyUSB0 flash monitor

0xjakob, 2022-01-20:
@gregrasche If you are on Linux (Ubuntu 20, in my case), then you should see an ACM interface coming up as soon as the ESP32-C3 is correctly powered up and in boot mode. You should be able to flash with
```
idf.py -p /dev/ttyACM0 flash monitor
```
This is under the condition that the ESP32-C3 is the only connected ACM interface device.

Regarding the hardware: if @sigmaaa's way doesn't work for you, then you can still change the resistors it's a bit tricky because they are 0402, but still doable. The nice thing is that you just need to change the orientation of these, since both positions have a common pad.

Here the resistor orientation for the usb-bridge:
[image]

And here a board after a mediocre solder job to change the orientation for the internal USB-Serial-JTAG device:
[image]

Note that the second picture is from a DevKitC, hence the differences of the surrounding traces/routing, but for the resistors, it doesn't matter.

Hope this helps.

gregrasche, 2022-01-20:
are these the pictures of a board that had the R21/R22 removed?

0xjakob, 2022-01-21:
> Do you have pictures of R21/R22 removal so that esp32c3 UART U0 is usable?

No, but according to the [PCB layout](https://dl.espressif.com/dl/schematics/PCB_ESP32-C3-DEVKITM-1_V1_20200915AA.pdf), R21 and R22 are the ones to the right of the circled area in the aforementioned pictures.

> Can you use the USB UART endpoint presented by the esp32c3 as the 'program interface', I think you said YES?.

Yes, you can program and read the ESP32-C3 using that USB-serial. I've done that many many times already. You may need to change the [default UART in menuconfig](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html?highlight=usb%20serial#config-esp-console-uart) though to route the UART output to the USB-serial interface instead of to UART0.

> Is there an APP NOTE or USER MANUAL on this subject?

For general usage yes: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html.
For changing the development board, I'm afraid no. Changing the resistors manually is not considered an actual use-case for the DEVKITM-1 development boards.

gregrasche, 2022-01-22:
You say
    Changing the resistors manually is not considered an actual use-case for the DEVKITM-1 development boards.

Having the resistors R1/R4 and R2/R3 in the alternate location would allow JTAG DEBUG over USB.  Why doesn't the board come with the resistors already moved to allow use of esp32c3 internal USB endpoints?

0xjakob, 2022-01-24:
@gregrasche The DEVKITM-1 board had been designed before the chip was available. Hence, it was not clear if the USB-serial functionality would work or not. I need to check if we sell development boards with the resistors changed. As already mentioned, you can always use the internal USB-serial and JTAG by just accessing GPIO18 and GPIO19.

0xjakob, 2022-01-24:
@gregrasche Currently, we do not sell development boards with USB-Serial and JTAG available via the micro-USB connector. We might sell some in the future but it is not determined yet.

closed 2022-01-27 by gregrasche