espressif/idf-eclipse-plugin #442: Functions highlights are not updated after an update of IDF

dizcza, 2022-01-19:
**Describe the bug**
I've manually updated the IDF to the latest version:
```
cd cd /media/dizcza/media/esp-idf/
git pull
./install.sh esp32
```

When I open the Eclipse, the function highlights show the function's description from an old version of IDF. For example, code highlights for the `esp_websocket_client_destroy` are showing the result for the `esp_websocket_client_send_bin` function. It must have cached an old version of the file - `esp_websocket_client.h` in this example - and I cannot update it to the newest version I've installed via terminal. I've tried Help > ESP-IDF Tools Manager > Install Tools with the path pointing out to the `/media/dizcza/media/esp-idf/` with no luck.

**Another issue is that there is no response when I click Help > Download and Configure ESP-IDF.** Is it broken?

**To Reproduce**
Assuming the Eclipse is already set up with an older version of ESP-IDF, steps to reproduce the behavior:
1. Update the ESP-IDF manually in a terminal.
2. Open the Eclipse
3. Ctrl-click any ESP-IDF function
4. It'll point to a line number of the same header file but of an older version.

**Expected behavior**
Ctrl-clicking to a function navigates to its definition.

**Environment Information:** 
[eclipse_configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/7895638/eclipse_configuration.txt)

**ESP-IDF Tools Information:**
```
Executing /home/dizcza/.espressif/python_env/idf5.0_py3.9_env/bin/python /media/dizcza/media/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)
```

**IDF Eclipse Plugin Version:** 
2.2.0.202108131352

**Eclipse Error log:** 
[error.log](https://github.com/espressif/idf-eclipse-plugin/files/7895659/error.log)

sigmaaa, 2022-01-19:
Hi @dizcza,
thanks for reporting this issue.
> Another issue is that there is no response when I click Help > Download and Configure ESP-IDF. Is it broken?

Yes, you are right, it's broken now. Fix for this is available in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/4975883491/artifacts/145208375).

Regarding the first issue, I will try to check/fix this bug and get back to you as soon as possible. Thanks

kolipakakondal, 2022-02-14:
Hello All, Fix is available as part of the IEP 2.4.1 release. Please get the latest changes using the update site. Thank you.

closed 2022-02-14 by kolipakakondal