espressif/idf-eclipse-plugin #446: "Download and Configure ESP-IDF" menu item fails

mukymuk, 2022-01-20:
**Describe the bug**

"Download and Configure ESP-IDF" menu item fails with an exception.

**To Reproduce**

Select "Espressif"->"Download and Configure" from the eclipse menu

**Environment Information:** 

[env.log](https://github.com/espressif/idf-eclipse-plugin/files/7903189/env.log)

**IDF Eclipse Plugin Version:** 

2.3.0.2021110916

**Eclipse Error log:** 

[error.log](https://github.com/espressif/idf-eclipse-plugin/files/7903219/error.log)

kolipakakondal, 2022-01-20:
Thanks for reporting @mukymuk 

This is fixed in the master branch and it will be part of v2.4.0 release. Most likely this will be available sometime next week.

ApplicationNinja, 2022-01-22:
So how can we use the plugin in now, while waiting for this fix?

ApplicationNinja, 2022-01-22:
> So how can we use the plugin in now, while waiting for this fix?

Never mind, I built the plugin locally...

Nnicholasnjue, 2022-01-24:
am also facing the same problem for a while now. any solution?

Nnicholasnjue, 2022-01-24:
> > So how can we use the plugin in now, while waiting for this fix?
> 
> Never mind, I built the plugin locally...

what do you mean by you built the plugin locally..?

Nnicholasnjue, 2022-01-24:
> > > So how can we use the plugin in now, while waiting for this fix?
> > 
> > 
> > Never mind, I built the plugin locally...
> 
> what do you mean by you built the plugin locally..?
am also facing the same problem for a while now. any solution?

# Executing C:\Users\ENGINEER\AppData\Local\Programs\Python\Python310\python.exe C:\Users\ENGINEER\esp-idf\tools\idf_tools.py list
Traceback (most recent call last):
  File "C:\Users\ENGINEER\esp-idf\tools\idf_tools.py", line 1675, in <module>
    main(sys.argv[1:])
  File "C:\Users\ENGINEER\esp-idf\tools\idf_tools.py", line 1671, in main
    action_func(args)
  File "C:\Users\ENGINEER\esp-idf\tools\idf_tools.py", line 1013, in action_list
    tools_info = load_tools_info()
  File "C:\Users\ENGINEER\esp-idf\tools\idf_tools.py", line 922, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\ENGINEER\\esp\\esp-idf;%IDF_PATH%\\tools;\\tools/tools.json'

IDF_PATH: C:/Users/ENGINEER/esp-idf/
IDF_PYTHON_ENV_PATH: /Users/user-name/.espressif/python_env/idf4.0_py3.7_env
PATH: C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microchip\xc8\v2.31\bin;C:\Users\ENGINEER\AppData\Local\Programs\Python\Python310;C:\Program Files\Git\cmd;C:\Users\ENGINEER\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\ENGINEER\AppData\Local\Programs\Python\Python310\;C:\Users\ENGINEER\AppData\Local\Microsoft\WindowsApps;;C:\Users\ENGINEER\AppData\Local\Programs\Microsoft VS Code\bin;/Users/user-name/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:/Users/user-name/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/Users/user-name/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:/Users/user-name/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/Users/user-name/.espressif/tools/ninja/1.9.0/:/Users/user-name/.espressif/python_env/idf4.0_py3.7_env/bin:/Users/user-name/esp/esp-idf/tools:$PATH

Operating System: windows 10
Java Runtime Version: 17.0.2+8-LTS-86
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.3.0.202110080828
ESP-IDF v4.3.1
'cmake' must be available on the PATH to use idf.py
Python set for IDF_PYTHON_ENV: Python 3.10.2
Python declared in PATH variable: Python 3.10.2

Nnicholasnjue, 2022-02-15:
> Hello All, Fix is available as part of the IEP 2.4.0 release. Please get
> the latest changes using the update site.

Hello, thanks for the update alert. I was so stressed with the
installation. but I will try it again.
"Hello All, Fix is available as part of the IEP 2.4.0 release"
The IEP 2.4.0 release you mention, do you refer the
https://github.com/espressif/idf-eclipse-plugin
as having the problem solved. I just download afresh and start the
installation or there is another link?

much appreciated
nicholas.

kolipakakondal, 2022-02-15:
Hi @Nnicholasnjue 

I understand. To address installation and configuration issues, we have released Espressif-IDE with v2.4.0. 

If you're a Windows user, this is much simpler now. We have released Espressif-IDE for Windows OS. You can find download links here https://dl.espressif.com/dl/esp-idf/

Our recent release
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.1

Documentation of Espressif-IDE
https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/EspressifIDE.md

Nnicholasnjue, 2022-02-16:
Thanks, I'll check it out.
after doing the complete installation. am stuck with the following error
when trying to create a new project

The console output looks like this:

*Installing tools...This can take a while. Please be patient.Executing
C:\Espressif\tools\idf-python\3.8.7\python.exe
C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py install
allTraceback (most recent call last):  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1828, in
<module>    main(sys.argv[1:])  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1824, in
main    action_func(args)  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1415, in
action_install    tools_info = load_tools_info()  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 942, in
load_tools_info    with open(tool_versions_file_name, 'r') as f:  # type:
ignoreFileNotFoundError: [Errno 2] No such file or directory:
'C:\\Users\\ENGINEER\\esp\\esp-idf;%IDF_PATH%\\tools;\\tools/tools.json'Executing
C:\Espressif\tools\idf-python\3.8.7\python.exe
C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py
install-python-envWARNING: Git was not foundWARNING: Is it not possible to
determine the IDF version: [Errno 2] No such file or directory:
'C:\\Users\\ENGINEER\\esp\\esp-idf;%IDF_PATH%\\tools;\\components\\esp_common\\include\\esp_idf_version.h'ERROR:
IDF version cannot be determinedExecuting
C:\Espressif\tools\idf-python\3.8.7\python.exe
C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py export
--format=key-valueTraceback (most recent call last):  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1828, in
<module>    main(sys.argv[1:])  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1824, in
main    action_func(args)  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1174, in
action_export    tools_info = load_tools_info()  File
"C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 942, in
load_tools_info    with open(tool_versions_file_name, 'r') as f:  # type:
ignoreFileNotFoundError: [Errno 2] No such file or directory:
'C:\\Users\\ENGINEER\\esp\\esp-idf;%IDF_PATH%\\tools;\\tools/tools.json'Configured
CDT Build Environment variables. You can check Preferences > C/C++ > Build
> EnvironmentConfigured CMake toolchain. You can check Preferences > C/C++
> CMakeInstall tools completed.*

Below I've attached a few pictures to demonstrate my problem.
kindly help.

best regards
Nicholas

kolipakakondal, 2022-02-16:
> Git was not found WARNING: Is it not possible to determine the IDF version: [Errno 2] No such file or directory

Seems to be tools installation didn't go smoothly because the GIT is not found. Could you check what is the path you are providing while installing tools: `Espressif > ESP-IDF Tools Manager > Install Tools`  and check `GIT executable location`