espressif/idf-eclipse-plugin #462: `size_t` fails to resolve - possible C misconfiguration

JAlexoid, 2022-01-29:
**Describe the bug**
Eclipse editor complains that `size_t` is unresolved.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubuntu 21.10
2. Install Python 3.10
3. Install OpenJDK 17
4. (follow basic install instructions)
5. Create a new project from `bluettooth/bluedroid/classic_bt/a2dp_sink`
6. Build project against ESP32
7. Refresh
8. Open file `bt_app_core.c` in Eclipse editor 

**Expected behavior**
C/CPP internal structures should be recognized and no error shown for `size_t` type.

**Screenshots**
[image]

**Environment Information:** 
[x.txt](https://github.com/espressif/idf-eclipse-plugin/files/7964832/x.txt)

**ESP-IDF Tools Information:**
Executing /home/alex/.espressif/python_env/idf4.4_py3.9_env/bin/python /home/alex/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

**IDF Eclipse Plugin Version:** 
Version: 2.4.0.202201280731

**Eclipse Error log:** 
No error related to this issue in Error Log.

kolipakakondal, 2022-02-17:
Hi @JAlexoid We could reproduce the issue. Will keep you posted on the fix.

ralphde, 2022-03-02:
On my side here it spits out malloc( (size_t) var ) as 'invalid argument' as well

hiperiondev, 2022-06-09:
I think it's because size_t is defined in more than one file (in my case I found 4 of them) and Eclipse can't decide which one is correct.

jmadsenee, 2022-06-21:
@hiperiondev Did you do something to eliminate the error?

hiperiondev, 2022-06-21:
> @hiperiondev Did you do something to eliminate the error?

No, the IDF plugin does not have the possibility to edit the files it indexes

jmadsenee, 2022-06-21:
In stddef.h, in the section that is about defining size_t, I found:
`#ifdef __BEOS__`
`typedef long ssize_t;`
`#endif /* __BEOS__ */`
So, for giggles, I changed the places in my code (not in the EMP32 code) that I was using size_t to ssize_t.  No more errors.

jmadsenee, 2022-06-22:
Well, so much for that...  seems size_t is an unsigned int and ssize_t is an int.
That makes me wonder why size_t is not showing an error in img_converters.h (in the esp32-camera driver)

kolipakakondal, 2022-07-27:
This seems to be working now not sure what did i do differently last time.
[image]

JAlexoid, 2022-08-17:
I did a clean install - IEP 2.6.0 Ubuntu 22.04

Still same problem