espressif/idf-eclipse-plugin #465: can not create launch configuration `ESP-IDF GDB OpenOCD Debugging`

kantrol, 2022-01-31:
On a fresh eclipse 2021-09 / idf-plugin 2.4 installation, creating a debug launch configuration is not possible.

**To Reproduce**

* New Launch Configuration ->
    ESP-IDF GDB OpenOCD Debugging -> Next
 
[image]

Even though the dialog can be edited, it is not possible to save (can not be closed with 'Finish' button)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
```
Executing /home/devecl2/.espressif/python_env/idf4.2_py3.9_env/bin/python /home/devecl2/develop/esp-idf-v4.2.2/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)
```
**IDF Eclipse Plugin Version:** 
2.4.0.2022012807

[eclipse_inst.txt](https://github.com/espressif/idf-eclipse-plugin/files/7971775/eclipse_inst.txt)

sigmaaa, 2022-02-01:
Hi @kantrol,

thank you for reporting this issue. Fix is available in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5133323085/artifacts/154584289). We will provide v2.4.1 with these changes maybe end of this week or early next week. 

The problem happens because of older openocd version, which does not support flashing over jtag, so you can install the provided  [build](https://github.com/espressif/idf-eclipse-plugin/suites/5133323085/artifacts/154584289) or update openocd or esp-idf:
to update openocd, please, refer to [this](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/JTAG%20Flashing.md#providing-the-right-path-for-openocd)

or you can use esp-idf 4.3 release and higher and then install tools. By doing it, you will get a version of the openocd  that supports jtag flashing

kantrol, 2022-02-01:
Oh, fast fix!
Works as expected now.
I just updated with the provided build.

Thank you, you probably can close this issue now.

kolipakakondal, 2022-02-14:
Hello All, Fix is available as part of the IEP 2.4.1 release. Please get the latest changes using the update site.

closed 2022-02-14 by kolipakakondal