espressif/idf-eclipse-plugin #474: Opening a terminal for the first time after Eclipse starts hangs at "Connecting..." and throws an NPE

gtjoseph, 2022-02-04:
**Describe the bug**
If you start Eclipse and try to open a Terminal to a connected device, the terminal will hang at "Connecting..." and an NPE will be thrown...
```
java.lang.NullPointerException: Cannot invoke "com.espressif.idf.serial.monitor.server.TerminalWebSocketServer.getPort()" because "com.espressif.idf.serial.monitor.server.SocketServerHandler.terminalWebSocketServer" is null
```
Full backtrace in the attached log.

If you close the terminal and open it again, it'll work.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a device via USB.
2. Start Eclipse and open an existing project.
3. Click the Terminal toolbar button.
4. Select the correct project and port and press OK

**Expected behavior**
The terminal _should_ open and the device should reset.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/8004891/eclipse_config.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
[espressif_tools.txt](https://github.com/espressif/idf-eclipse-plugin/files/8004909/espressif_tools.txt)

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
```
  Espressif IDF Plugins for Eclipse	2.4.0.202201280731	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD
```
For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon
[image]

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[eclipse_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/8004896/eclipse_log.txt)

sigmaaa, 2022-02-07:
Hi @gtjoseph,

Thank you for reporting this issue. Looks like this issue happens, when I'm trying to open the terminal with GDBStub on panic mode. Fix for this is available in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5208317371/artifacts/159224954). 
We will provide the v2.4.1 release soon, probably with this fix. I will update this topic again when these changes will be in stable release

gtjoseph, 2022-02-08:
Thanks!  Will test shortly.

sigmaaa, 2022-02-11:
ESP-IDF Eclipse Plugin v2.4.1 is published [here](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.1). It includes a fix for this issue

gtjoseph, 2022-02-11:
Built 2.4.1 from git and installed and the terminal now works the first time.
Thanks for the quick turn around!!

closed 2022-02-11 by gtjoseph