espressif/idf-eclipse-plugin #479: esp32c3 camera recommendations

gregrasche, 2022-02-11:
We are looking at usage of the esp32c3 SOC, and would like to have a fairly fast camera interface.   Let's say a camera that can take an image once per second. 

Questions:  
1) Do you have a vendor that you have worked with who has a 4 M pixel (or greater) camera that can easily interface to the ESP?
2) What ESP interface (SPI, DVP, I2S, or other) is used?
3) Do you have a SW package to manage the camera?

kolipakakondal, 2022-02-11:
Hi @gregrasche As this is not an eclipse plugin question, would you mind posting this question in [esp32 forum](https://www.esp32.com/viewforum.php?f=2) to get a better response.

kolipakakondal, 2022-02-14:
https://www.esp32.com/viewtopic.php?f=2&t=26067

closed 2022-02-14 by kolipakakondal