espressif/idf-eclipse-plugin #480: openOCD can not launch debug session but can flash using JTag

ralphde, 2022-02-13:
after installing a complete fresh environment , as a simple update didn't work at all, I remain with the problem that any debug launch configuration can not start openOCD. However I can flash using Jtag.

Noteable as well, when I try to launch a debug session I can not flash after as well. I have to unplug the ESP-Prog and plug in again.

here the error log of the intent to flash the device after tried a debug session

 
[error2.log](https://github.com/espressif/idf-eclipse-plugin/files/8056416/error2.log)
 after this error log was created I replugged the USB and the device flashed seamlessly. I also can flash many times with no issue. 

 here the flash attempt failed again after debug attempt
[error3.log](https://github.com/espressif/idf-eclipse-plugin/files/8056426/error3.log)

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
`Executing C:\.esp-tools\python_env\idf4.4_py3.10_env\Scripts\python.exe C:\.esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

`
**IDF Eclipse Plugin Version:** 
ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD	ESP-IDF Eclipse Plugin	2.4.1.202202101602	com.espressif.idf.feature

**Eclipse Error log:** 
```
!ENTRY org.eclipse.cdt.dsf 4 10005 2022-02-13 19:30:53.397
!MESSAGE Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.concurrent.RequestMonitorWithProgress@443bd649): Status ERROR: ilg.gnumcueclipse.debug.gdbjtag code=5010 Starting OpenOCD timed out. null' resulted in an error.
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2022-02-13 19:30:53.397
!MESSAGE Starting OpenOCD timed out.

!ENTRY org.eclipse.cdt.dsf.gdb 4 5012 2022-02-13 19:30:53.401
!MESSAGE Error in services launch sequence
!STACK 1
org.eclipse.core.runtime.CoreException: Starting OpenOCD timed out.
	at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
	at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:831)
!SUBENTRY 1 ilg.gnumcueclipse.debug.gdbjtag 4 5010 2022-02-13 19:30:53.401
!MESSAGE Starting OpenOCD timed out.

!ENTRY com.espressif.idf.core 1 0 2022-02-13 19:30:53.453
!MESSAGE Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

!ENTRY com.espressif.idf.debug.gdbjtag.openocd 4 1 2022-02-13 19:30:53.455
!MESSAGE C:\.esp-tools\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/openocd.exe -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\.esp-tools\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-1.8v.cfg -c program_esp_bins C:/Users/ralph/.workspace20220213/maxill/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"

!ENTRY org.eclipse.core.jobs 4 2 2022-02-13 19:30:53.609
!MESSAGE An internal error occurred during: "Starting OpenOCD".
!STACK 0
java.util.concurrent.RejectedExecutionException: Task java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask@b729a05[Not completed, task = java.util.concurrent.Executors$RunnableAdapter@3a82c554[Wrapped task = ilg.gnumcueclipse.debug.gdbjtag.dsf.GnuMcuGdbServerBackend$5@eb3e80b]] rejected from org.eclipse.cdt.dsf.concurrent.DefaultDsfExecutor@78e09b60[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 41]
	at java.base/java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2057)
	at java.base/java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:827)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor.delayedExecute(ScheduledThreadPoolExecutor.java:340)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor.schedule(ScheduledThreadPoolExecutor.java:562)
	at org.eclipse.cdt.dsf.concurrent.DefaultDsfExecutor.schedule(DefaultDsfExecutor.java:450)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor.submit(ScheduledThreadPoolExecutor.java:715)
	at org.eclipse.cdt.dsf.concurrent.DefaultDsfExecutor.submit(DefaultDsfExecutor.java:482)
	at ilg.gnumcueclipse.debug.gdbjtag.dsf.GnuMcuGdbServerBackend.checkServer(GnuMcuGdbServerBackend.java:459)
	at ilg.gnumcueclipse.debug.gdbjtag.dsf.GnuMcuGdbServerBackend$GdbServerStep$2.run(GnuMcuGdbServerBackend.java:722)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

`

`
[eclipse configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/8056398/eclipse.configuration.txt)
`

DigKleppe, 2022-02-14:
Hi, i am struggling with the same issue,
The only way i can get programming and debugging by jtag ok is to use GDB OpenOCD debugging and program the binary with the openOCD commandline:
-s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg  -c "program_esp  ${workspace_loc:/wrover/build/app-template.bin} 0x10000 verify "

The other debugging items do not work,  are leaving my poor esp32 in a hopeless state.
Can someone explain these?

gaupen1186, 2022-02-15:
I have the same issue too !
And I have to fall back to v2.4.0 now.
I'm curious how this happened?
No regression testing?

sigmaaa, 2022-02-15:
Hi all,

thank you for reporting this issue. I think the problem may be in Espressif gdb server launch timeout preference. This preference is set for me, so I can't reproduce this issue, but since you have this issue, most likely there is some mistake in getting the default value for this preference. 

Please, try to do next:
- Open eclipse, then go to Window -> Preferences -> Espressif -> you will see a page like on the screenshot below:
[image]

- On this page, check if the value for timeout is set and click on "Apply and Close" or "Apply" even if this value was there.

try to debug again and let me know if it helped. Thanks

ralphde, 2022-02-15:
I checked this page and applied and could not experience any change

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke

sigmaaa, 2022-02-15:
Hi @ralphde,

can you try this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5317387730/artifacts/165470357), please? Or you can try to use [Espressif-IDE](https://github.com/espressif/idf-eclipse-plugin/suites/5317387730/artifacts/165470361) which also has this build. In this build, I changed the way we are getting a default value for openocd timeout.  I tried to install this build in a fresh environment and there is no such problem either, so I hope it will help. Thank you

ralphde, 2022-02-16:
When I start the IDE a windows cmd pops up with this message
[image: image.png]

can I trust this setup despite this message?

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke

sigmaaa, 2022-02-16:
Hi @ralphde,

Whenever eclipse crashes, it will generate `hs_err_pidxx.log` kind of file. By default system creates it in the working directory of the process. Could you please provide it so we can analyze it further? Thank you

ralphde, 2022-02-16:
Such a logfile does not exist in my installation, however the throughn
error is the one explained here.

http://www.slf4j.org/codes.html#StaticLoggerBinder

So Java is not at all my baby, but this looks to me like connected with
creating these log files. So may be this error is through, because the
error handler tries to log the event and fails.

Sorry can't deliver more

Ralph Deffke

DigKleppe, 2022-02-16:
My issue starting the debugger is fixed by setting the timeout to 50 seconds. So i can use the debugger the easy way.
Another question: does the (default) elf file contain all segments, including spiffs?
Dig

ralphde, 2022-02-17:
@sigmaaa  so I tried the plugin 
[image]
 you provided and the problem persists!

 I can flash using JTag but flashing as part of the a debug launch fails
openOCD 0.11

here the console output
`
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Flashing C:/esp-ws-202109-17/blink/build/partition_table/partition-table.bin at 0x8000
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu1: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : target esp32.cpu0 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu1: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Couldn't halt target before SoC reset
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet
Error executing event gdb-attach on target esp32.cpu0:
embedded:startup.tcl:530: Error: 
in procedure 'program_esp' called at file "C:\Users\ralph\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 220
in procedure 'program_error' called at file "C:\Users\ralph\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 93
at file "embedded:startup.tcl", line 530
Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
`

in reverse the console running a run launch which works seamlessly 

`Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Flashing C:/esp-ws-202109-17/blink/build/partition_table/partition-table.bin at 0x8000
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s2 on 3333
Info : Listening on port 3333 for gdb connections
[... 62 lines cut ...]
Info : esp32s2: Target halted, PC=0x4000F988, debug_reason=00000000
Info : esp32s2: Core was reset.
shutdown command invoked`

to mention is that I use the blink example and flashing via the provided run configuration using Jtag works.
However, it is a miracle for the user where the settings for openOCD come from in the provided launch run configuration.

because if I want to define my own run only configuration I am ask to enter the debugger details just like I would define a debug launch.
[image]

while when I edit the provided run configuration it looks like this

[image]

So why do I not get this window when i wanna do just a run configuration?

AND I ask this now many times WHERE ARE ${openocd_path} AND ${openocd_executable} DEFINED ? please answer and let us know where we should define those. Workspace or project? or are we free and these variables are as well not set anywhere
${cross_prefix}gdb${cross_suffix}

There are more discrepancies but these are the main once.

Please keep on with this I am in the middle of a commercial project and need this new version with isolated components, which is the best thing you ever came up with.

thanks
Ralph

sigmaaa, 2022-02-17:
Hi @ralphde

Looking at your first screenshot, I would guess that the `OPENOCD_SCRIPTS` variable is missing, which is why you don't see the options as you see on the second screenshot. But it's very strange that you see these options on the second screenshot without the `OPENOCD_SCRIPTS` variable, because code check for these two tabs is the same. That's how the launch configuration looks like if something wrong with openocd variable:
[image]

Let's look at the problems one by one:
-  ${openocd_path} and ${openocd_executable} are defined at Preferences -> Espressif -> Global OpenOCD Path
[image]

NOTE: both folder and executable are set automatically based on the `OPENOCD_SCRIPTS` variable. So to check if this variable is set:
- Preferences -> C/C++ -> Build -> Environment -> after setting `OPENOCD_SCRIPTS` restart eclipse ->  Global OpenOCD Path should be configured automatically 
[image]

What else you can do is:

- try to reinstall tools
- Try to export error log, after you edit debug configuration and start the debug session. To do it, please follow [these steps](https://github.com/espressif/idf-eclipse-plugin#error-log).

ralphde, 2022-02-17:
@sigmaaa Look please don't ask for a reinstall ever again, it comes barefaced here. I tried a complete fresh install using the extension you provided earlier in this conversation.

Fact is the thing does not work out of the box.

The variables you mentioned are perfectly set in my environment:
[image]

and

[image]
 
So clearly these variables are not set by pulling from here when I try to do my own run only configuration.

[image]

However, when I try a ESP debug session these variables seem to be set:
[image]

But again even if the path is set and correct, I can not flash and debug. See above console output

ralphde, 2022-02-17:
@sigmaaa I thing the ESP Extension should set these variables correct so that you can find them opening the available variable list. 

Also I am asking why do I not have the same screen for a run only configuration as the extension itself provides?

And main problem please why can I not flash/debug with these, obviously correct, settings?

I need to get that fixed

ralphde, 2022-02-17:
@sigmaaa another think I find suspicious is this
[image]
 
so the blink configuration here is the original provided one, and the icon tell me that this is a GDB openOCD debugging based configuration. As you can see the ISP-EDF debug configuration has a different icon
[image]

May I ask for more details on this and what is the difference? Why does the provided work and which debugger is used and why does ANY other configuration not work?

sigmaaa, 2022-02-17:
@ralphde now I see where the problem is. I haven't noticed, that you were using GDB OpenOCD Debugging instead of ESP-IDF GDB OpenOCD Debugging on your screenshot. Sorry for that misunderstanding.

GDB OpenOCD Debugging it's an external plugin that comes from Eclipse CDT.
ESP-IDF GDB OpenOCD Debugging is part of our plugins. Please refer to this [GDB OpenOCD Debugging](https://github.com/espressif/idf-eclipse-plugin/tree/master/docs/OpenOCD%20Debugging.md)
The other option to debug your app is GDB hardware debugging. Please refer to [GDB Hardware Debugging guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/index.html)
Answering to your question regarding ${openocd_path} and ${openocd_executable}. For GDB OpenOCD Debugging plugin 
are stored under MCU 
[image]

To debug your project with our plugin, you need to select debug mode and configuration with the ESP-IDF GDB OpenOCD Debugging icon, which is in your case `blink configuration`. 

> May I ask for more details on this and what is the difference? Why does the provided work and which debugger is used and why does ANY other configuration not work?

Do I understand correctly, that provided configuration, which is `blink Configuration` works for you, but `launch run blink` doesn't?

ralphde, 2022-02-17:
no sorry, not one is working !!!!! except the provided one when I do new
sample project

ralphde, 2022-02-18:
@sigmaaa I got it working now, I don#t know what fixed the problem but i can tell what I did. Following your advice I reviewed all settings 
I opened Preferences -> C/C++ -> Build -> Environment  checked (was fine) and saved
and what I did as well (and didn't before as I was not aware of this) 
I opened Preferences -> MCU -> Global OpenOCD Path, checked (was OK) and saved

restart and it worked

I can debug now with one little problem regarding uxTopUsedPriority is not defined, however I will open a new ticket about that.

But the fact that this installation does not work out of the box remains.

closed 2022-09-30 by sigmaaa