espressif/idf-eclipse-plugin #481: Panic handler behaviour (GDBStub on panic) no longer starts gdb in terminal

gtjoseph, 2022-02-13:
**Describe the bug**
Prior to 2.4, if you had a terminal window connected to a device and it paniced, gdb would automatically started in the terminal window when "Panic handler behaviour" was set to "(GDBStub on panic)".   Unfortunately now, the terminal shows the following:
```
Entering gdb stub now.
$T0b#e6Communicating through WebSocket
WebSocket sent: {'event': 'gdb_stub', 'port': '/dev/ttyUSBESP-S3-002', 'prog': '/usr/src/mcu/esp32/projects/wear_levelling/build/wear_levelling.elf'}
Waiting for debug finished event
```
Then the launch bar in the IDE changes to...
[image]
If you click the "Launch in Debug mode" button, you eventually get "Timed out trying to launch GDB".

If you run "idf monitor" from a terminal window outside Eclipse, gdb _is_ automatically started.

**To Reproduce**
Steps to reproduce the behavior:
1. Create any simple IDF project.
2. In menuconfig, set "Panic handler behaviour" to "(GDBStub on panic)".
3. Just inside the app_main function add "abort();"
4. Open a terminal from the Eclipse toolbar and connect to the device.
5. Flash.
6. When the device restarts, it will show the above behavior.

**Expected behavior**
When the device restarts and the "abort" is hit, you should get something like:
```
Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/src/mcu/esp32/projects/wear_levelling/build/wear_levelling.elf...
Remote debugging using /dev/ttyUSBESP-S3-002
0x403757a9 in esp_backtrace_print_from_frame (depth=1070548651, frame=0x3fcf46d8, panic=10) at /usr/src/mcu/esp32/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:76
76	    print_str("\r\n\r\nBacktrace:", panic);
(gdb)
```

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/8056619/eclipse_config.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
[installed_tools.txt](https://github.com/espressif/idf-eclipse-plugin/files/8056624/installed_tools.txt)

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon
[image]

**Eclipse Error log:** 
Window > Show View > Error Log
[error_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/8056638/error_log.txt)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

kolipakakondal, 2022-02-16:
Hi @gtjoseph 

Sorry for the late reply. This could be due to the GDB server timeout issue reported by other users. It's basically trying to start the OpenOCD server and getting to GDBStub debugging mode (as GDBStub on panic is enabled) but due to issues on GDB server timeout, it's unable to proceed further.

Will get back to you on this asap.

kolipakakondal, 2022-02-16:
Hi @gtjoseph Could you please check our latest dev build which has a fix for the GDB server timeout issue?

Steps:
1. Download latest dev build plugin from [here]( https://github.com/espressif/idf-eclipse-plugin/suites/5331618932/artifacts/166313754)
2. Follow the local archive site installation instructions from [here](https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile) to install it

gtjoseph, 2022-02-16:
No change. 
```
org.eclipse.core.runtime.CoreException: Timed out trying to launch GDB.
	at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
	at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:831)
```

Exactly what's supposed to happen?  The same behavior as before where gdb automatically starts in the terminal or something else?

gtjoseph, 2022-02-16:
Not sure of this means anything but if I copy the command from the gdbstub_debug_launch configuration and run it from a shell, it works fine...

```
sdspi ᐅ /home/gtjoseph/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gdb -ex "set serial baud 115200" -ex "target remote /dev/ttyUSB2" -ex interrupt "/usr/src/mcu/esp32/projects/sdspi/build/sdspi.elf"
GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/src/mcu/esp32/projects/sdspi/build/sdspi.elf...
Remote debugging using /dev/ttyUSB2
0x403757a9 in panic_abort (details=0x3fcf46ab "abort() was called at PC 0x420054c2 on core 0") at /usr/src/mcu/esp32/esp-idf/components/esp_system/panic.c:402
402	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
(gdb) bt
#0  0x403757a9 in panic_abort (details=0x3fcf46ab "abort() was called at PC 0x420054c2 on core 0") at /usr/src/mcu/esp32/esp-idf/components/esp_system/panic.c:402
#1  0x40378bcc in esp_system_abort (details=0x3fcf46ab "abort() was called at PC 0x420054c2 on core 0") at /usr/src/mcu/esp32/esp-idf/components/esp_system/esp_system.c:128
#2  0x4037e859 in abort () at /usr/src/mcu/esp32/esp-idf/components/newlib/abort.c:46
#3  0x420054c5 in app_main () at ../main/sd_card_example_main.c:67
#4  0x4201757c in main_task (args=0x0) at /usr/src/mcu/esp32/esp-idf/components/freertos/port/port_common.c:129
#5  0x4037b6d4 in vPortTaskWrapper (pxCode=0x42017508 <main_task>, pvParameters=0x0) at /usr/src/mcu/esp32/esp-idf/components/freertos/port/xtensa/port.c:131
(gdb) 
```

kolipakakondal, 2022-02-17:
Similar to what happened in the command line it should connect to the debugger in the IDE and stop at the point where the exception has occurred so that you should be able to visualize the current state of the thread, variables, registers, etc

alirana01, 2022-02-17:
@gtjoseph 
Can you try to increase the timeout for the GDB Server
[image]

gtjoseph, 2022-02-17:
I changed it to 60 but it didn't help.  
I just noticed that after the terminal prints...
```
Entering gdb stub now.
$T0b#e6Communicating through WebSocket
WebSocket sent: {'event': 'gdb_stub', 'port': '/dev/ttyUSB2', 'prog': '/usr/src/mcu/esp32/projects/sdspi/build/sdspi.elf'}
Waiting for debug finished event
```
the terminal closes.
[image]

Shouldn't it stay open?

alirana01, 2022-02-17:
@gtjoseph So what happens is that when you have selected that GDBStub on Panic option we look for the gdb_stub event from the terminal on socket server and when that event has occurred the terminal connection is closed because the debugger is now attached to the device's gdb stub 
You can try to switch the perspective to debug and you may see that the debugger is showing the info like threads stacktrace and variables.

Can you try to switch the perspective to debug and see if you are getting the board info related to panic?

gtjoseph, 2022-02-17:
This is what I did...
* Created a new IDF project from examples/storage/sd_card/sdspi.
* Did an `idf set-target esp32s3` from the command line.
* Added an `abort()` statement just after `app_main()` starts.
* Create a new target for esp32s3 and serial port /dev/ttyUSB2.
[image]

* Switched to the Debug perspective.

* Clicked the "Build" button...
[image]
`Build complete (0 errors, 0 warnings): /usr/src/mcu/esp32/projects/sdspi/build`

* Clicked the "Launch in 'Run' mode" button to flash.
```
Wrote 187616 bytes (98897 compressed) at 0x00010000 in 3.0 seconds (effective 499.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 525.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

* Nothing else happened so I opened a terminal ...

[image] 
and got...
```
I (0) cpu_start: Starting scheduler on APP CPU.
abort() was called at PC 0x420054c2 on core 0
0x420054c2: app_main at /usr/src/mcu/esp32/projects/sdspi/build/../main/sd_card_example_main.c:67
Backtrace:0x403757a6:0x3fcf46600x40378bc9:0x3fcf4680 0x4037e856:0x3fcf46a0 0x420054c2:0x3fcf4710 0x42017579:0x3fcf4730 0x4037b6d1:0x3fcf4750 
0x403757a6: panic_abort at /usr/src/mcu/esp32/esp-idf/components/esp_system/panic.c:402
0x40378bc9: esp_system_abort at /usr/src/mcu/esp32/esp-idf/components/esp_system/esp_system.c:128
0x4037e856: abort at /usr/src/mcu/esp32/esp-idf/components/newlib/abort.c:46
0x420054c2: app_main at /usr/src/mcu/esp32/projects/sdspi/build/../main/sd_card_example_main.c:67
0x42017579: main_task at /usr/src/mcu/esp32/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)
0x4037b6d1: vPortTaskWrapper at /usr/src/mcu/esp32/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: c4c6142ccecfacb4
Entering gdb stub now.
$T0b#e6Communicating through WebSocket
WebSocket sent: {'event': 'gdb_stub', 'port': '/dev/ttyUSB2', 'prog': '/usr/src/mcu/esp32/projects/sdspi/build/sdspi.elf'}
Waiting for debug finished event
```
The launch bar changed to...
[image]
but the debugger didn't start by itself...
[image]

* So I clicked the "Launch in 'Debug' mode" button.

[image]
and got...
```
null
org.eclipse.cdt.dsf.gdb
Error
Thu Feb 17 06:41:43 MST 2022
Error in services launch sequence

org.eclipse.core.runtime.CoreException: Timed out trying to launch GDB.
	at org.eclipse.cdt.dsf.concurrent.Sequence.abortExecution(Sequence.java:599)
	at org.eclipse.cdt.dsf.concurrent.Sequence$2.handleErrorOrWarning(Sequence.java:448)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleFailure(RequestMonitor.java:433)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor.handleCompleted(RequestMonitor.java:393)
	at org.eclipse.cdt.dsf.concurrent.RequestMonitor$2.run(RequestMonitor.java:317)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:831)

null
org.eclipse.cdt.dsf
Error
Thu Feb 17 06:41:43 MST 2022
Request for monitor: 'RequestMonitor (org.eclipse.cdt.dsf.concurrent.RequestMonitorWithProgress@5a43c291): Status ERROR: org.eclipse.cdt.dsf.gdb code=5010 Timed out trying to launch GDB.' resulted in an error.

```

alirana01, 2022-02-25:
@gtjoseph I am not able to reproduce the issue locally can we connect on teams to see what the issue is and how I can help you out with it? You can email me a time suitable for you and I will try to get in touch at that time.

Thanks
ali.azamrana@espressif.com

gtjoseph, 2022-02-25:
Ah, no worries.  If you can't reproduce it, it's probably something in my configuration.  I'm going to start with a fresh esp-idf, eclipse and plugin and see if it still happens.  If it does, I can add some extra debugging output to the plugin and see if I can narrow down what's going on.

gtjoseph, 2022-02-25:
Just FYI... Still have the issue with clean eclipse cdt, esp-idf, and plugin installs.  Could this be related to using the eclipse 2021-12 build instead of the older 2021-09 build?

alirana01, 2022-03-08:
Hi @gtjoseph can you please try it with the older version of eclipse-2021-09?

And if it doesnt work just drop me an email ali.azamrana@espressif.com I will be happy to schedule a call and see what I can help you with

gtjoseph, 2022-03-08:
So, I went back to 2021-09 but staying with the current esp-idf 4.4 and the latest plugin and had the same issue.   I then went back to the last commit of esp-idf in 2021, reinstalled the tools, etc. and it worked!   So then I went back to current esp-idf and as expected, it failed.  Back again to the last commit of 2021, and it still failed.  So no bisecting for me. :)

Anyway, one thing I did notice since I was starting eclipse from a bash shell was that when when gdb was supposed to start in the monitor window, eclipse spit out `exec_pty(71): returning due to error: Bad file descriptor` in the bash terminal.  I'd never seen that in any of the logs so maybe that tells you something.

> And if it doesnt work just drop me an email [ali.azamrana@espressif.com](mailto:ali.azamrana@espressif.com) I will be happy to schedule a call and see what I can help you with

I really appreciate the offer but I can't really spend any more time on this.  I've changed my workflow to do a build in eclipse but run `idf.py flash  monitor` directly from a bash shell.   When I'm debugging live, it's a lot easier to just run openocd in one window and the monitor in another without eclipse getting in the way.

Thanks again!
If the `bad file descriptor` message tells you something, great.  Otherwise I'm fine with closing this issue.

alirana01, 2022-03-09:
@gtjoseph Thanks for the detailed description and your patience I will try to reproduce using your steps and will close the issue afterwards.

alirana01, 2022-09-30:
Internal ticket created for further investigation on this and the core dump issue on Linux.

closed 2022-09-30 by alirana01