espressif/idf-eclipse-plugin #483: esp32c3 build "java null pointer"

jiauka, 2022-02-15:
Both stable/mastewr and beta releases have this problem when building.

	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:392)

On a clean UBUNTU 20.04 and Eclipse 2021 09 installation. It also happens with Eclipse 2021-06 and the latest eclipse cpp

full stack error
org.eclipse.core.runtime.CoreException: Building blink
	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:392)
	at org.eclipse.cdt.core.build.CBuilder.build(CBuilder.java:45)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:853)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:232)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:281)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:334)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:337)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:296)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:488)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:565)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:603)
	at org.eclipse.core.internal.resources.Project.build(Project.java:116)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.lambda$0(LaunchConfigurationDelegate.java:406)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2333)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildProjects(LaunchConfigurationDelegate.java:412)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildForLaunch(LaunchConfigurationDelegate.java:122)
	at org.eclipse.launchbar.core.target.launch.LaunchConfigurationTargetedDelegate.superBuildForLaunch(LaunchConfigurationTargetedDelegate.java:61)
	at com.espressif.idf.launch.serial.internal.SerialFlashLaunchConfigDelegate.buildForLaunch(SerialFlashLaunchConfigDelegate.java:249)
	at org.eclipse.launchbar.core.target.launch.LaunchConfigurationTargetedDelegate.buildForLaunch(LaunchConfigurationTargetedDelegate.java:50)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:777)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:716)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1021)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1224)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.lang.NullPointerException
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1092)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
	at com.espressif.idf.core.util.ParitionSizeHandler.startProcess(ParitionSizeHandler.java:66)
	at com.espressif.idf.core.util.ParitionSizeHandler.startIdfSizeProcess(ParitionSizeHandler.java:78)
	at com.espressif.idf.core.util.ParitionSizeHandler.startCheckingSize(ParitionSizeHandler.java:42)
	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:383)
	... 27 more

sigmaaa, 2022-02-15:
Hi @jiauka,

Thank you for reporting this.
I can't reproduce this in my environment. Can I ask you to click on `Espressif -> Product information` and share the output here? Thank you

jiauka, 2022-02-17:
And yes. I'm using v2.4.1. It also happens with the latest master branch.

It ONLY happens building for the esp32c3. Building for the ESP32 works fine.
I've tried many combinations:
ECLIPSE 2021-06, 2021-09 and 2021-12
IDF plugins: 2.4.0, 2.4.1 and master branch.
Installing only espc3 tools or all tools, and so.

Building from command line works fine under all combinations, using eclipse is a no go.
I even setup a "clean" xubuntu 20.04 virtual machine in order to do a fresh install. Same problem. It is able to do a build the very first time, but once I restart eclipse it is no longer able to build any esp32c3 app, neither an examples, nor a self made. No matter I open them or I import them from eclipse. 

Espressif -> Product information

`Executing /home/jaume/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/jaume/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

IDF_PATH: /home/jaume/esp/esp-idf
IDF_PYTHON_ENV_PATH: <NOT FOUND>
PATH: /home/jaume/.espressif/tools/xtensa-clang/12.0.1-d9341b81fc/xtensa-esp32-elf-clang/bin:/home/jaume/.espressif/tools/cmake/3.20.3/bin:/home/jaume/.espressif/tools/ninja/1.10.2/:/home/jaume/esp/esp-idf/components/esptool_py/esptool:/home/jaume/esp/esp-idf/components/espcoredump:/home/jaume/esp/esp-idf/components/partition_table:/home/jaume/esp/esp-idf/components/app_update:/home/jaume/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/home/jaume/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/home/jaume/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/jaume/.espressif/python_env/idf4.4_py3.8_env/bin:/home/jaume/esp/esp-idf/tools:/home/jaume/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

Operating System: linux
Java Runtime Version: 16.0.2+7-67
Eclipse Version: 4.21.0.v20210906-0500
Eclipse CDT Version: 10.4.0.202106130052
IDF Eclipse Plugin Version: 2.4.1.202202101602
ESP-IDF v4.4
Python set for IDF_PYTHON_ENV: Python 3.8.10
Python declared in PATH variable: Python 3.8.10
`

sigmaaa, 2022-02-17:
Hi @jiauka,

I have the same product information log, only java version is different. Also tried different linux (mint and ubuntu 20.04), but unfortunately, I am still not able to reproduce this issue. Can I ask you please to try to build the project again on 2.4.1 and export the complete error log here? There can be some other issue before the build, which leads to the build problem. Thank you

sigmaaa, 2022-02-18:
Hi @jiauka,

what I found strange in your product info log is the `IDF_PYTHON_ENV_PATH:` variable, it shouldn't be empty. Even if this value is null, the product information log will print it like `IDF_PYTHON_ENV_PATH: <NOT FOUND>`. 
Also I see this env path in PATH variable /home/jaume/.espressif/python_env/idf4.4_py3.8_env/bin. We are getting this variable if IDF_PYTHON_ENV_PATH is null, so it should be ok, but it's not for some reason. Also, I see, that python is executed correctly to show the tools list and python version. 

Even if everything looks ok, the root of the problem may be in getting python from python_env sometimes, because we use it to check the remaining partition size:

> at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1092)
> at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
> at com.espressif.idf.core.util.ParitionSizeHandler.startProcess(ParitionSizeHandler.java:66)

can you please check if `IDF_PYTHON_ENV_PATH` is ok? 

- Eclipse -> C/C++ -> Build -> Environment -> IDF_PYTHON_ENV_PATH

But even if it's problem with `IDF_PYTHON_ENV_PATH` I still don't understand how it can happen only on esp32c3 target, we are starting this process for all targets. 

Also, what can help us to find the root of the problem is the console output. Can you see something similar after the build in the console? 
[image]

sigmaaa, 2022-04-05:
Hi @jiauka,

did you manage to solve the problem? If not, you can try[ 2.4.2 build](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.2), it has some changes, that might be related to your issue. Thanks

jiauka, 2022-04-06:
HI:
I have moved to the master branch, and with the 2.4.1 plugin it more or less works. Usually I have to start the debug twice to work.
I also tried Visual Studio code and I'm totally unable to debug my software. A simple blink or alike works, but as soon as there are several task I can no step on the debug under VSCode. I will open a new ticket for that.

closed 2022-04-13 by sigmaaa