espressif/idf-eclipse-plugin #485: Version 2.4.1 (Eclipse) and 2.4.0 (Espressif IDE): Build fail after image is build for working project without clear reason.

Maksons, 2022-02-16:
After update plugin to version 2.4.1 build fail at the end when image is generated (compilation without errors).
Tried with Espressif IDE also and got same result.

[image]

Example project "Hello word" builds without problems.

Maksons, 2022-02-16:
More detailde investigation in .log file show the problem source:
    - compiled binary filename is:  app-template.bin
    - but final stage looking for binary name: TTS_BigTarget.bin

Maksons, 2022-02-16:
Problem resolved by fixing top level CMakeLists.txt
`
project(app-template)
`
replaced with :
`
get_filename_component(ProjectId ${CMAKE_CURRENT_LIST_DIR} NAME)
string(REPLACE " " "_" ProjectId ${ProjectId})
project(${ProjectId})
`

sigmaaa, 2022-02-16:
Hi @Maksons,

Thank you for reporting this issue. Check if CMakeLists.txt under your project folder contains these lines:
`get_filename_component(ProjectId ${CMAKE_CURRENT_LIST_DIR} NAME) string(REPLACE " " "_" ProjectId ${ProjectId}) project(${ProjectId})`

with these two lines, the project binaries will match the project name.

ammaree, 2022-02-16:
I think the problem, and possibly solution, is more complex than this.
Our build system generate ~20 differently named ??????.bin/elf/map files
- all from the exact same source code in the exact same Eclipse project folder,
- based on the specific hardware platform selected and the functional components included,
- finish with a build "error" message at the end of a successful build,
- but with the correctly named output files in the ?/build subdirectory.

The error message is always, independent of the options selected, as follows:
[image]

Seems the plugin is looking for irmacs.bin (irmacs being the name of the project folder) but the output files will always be named differently, based on the ${ProjectId} variable. This error also results in us not being able to flash from the IDE since it cannot wind the irmacs.??? files.

sigmaaa, 2022-02-16:
Hi @ammaree,

in this [build](https://github.com/espressif/idf-eclipse-plugin/suites/5328309873/artifacts/166118710) I changed the way we search for the bin file during the build. Now we get the binaries based on "app_bin" in project_description.json which is in the build folder. Thus, with these changes, there is no need to make changes to the CMakeLists.txt file mentioned above.

ammaree, 2022-02-16:
@sigmaaa 

Thanks, but what do I do with the build to apply it and fix my problem.
Or better to wait for the next release (2.4.2) coming when?

sigmaaa, 2022-02-16:
@ammaree 
 
 if you want to install this build to take a look if this fixes a problem for your project, you need to follow steps mentioned [here](https://github.com/espressif/idf-eclipse-plugin#installing-idf-eclipse-plugin-from-local-archive). 
 Other option is to download Espressif-IDE with this build and test it there. Looks like you are using mac, here is [espressif-ide-macos](https://github.com/espressif/idf-eclipse-plugin/suites/5328309873/artifacts/166118712).
 
 Also, I will update this topic, when PR with these changes will be approved and be included to the next release. Thank you

ammaree, 2022-02-16:
@sigmaaa 

Thanks, have used the installer option. The build itself still works with correct output files names and no error at the end.

However, it now starts with an error message as below.
[image]

Subsequently this error stop "Launch" from running...

sigmaaa, 2022-02-16:
@ammaree 

'Animation start' the problem may be due to `-Xdock:icon` option in eclipse.ini points to a non-existing .icns file. 

how to check it on mac: 

- Right click on eclipse icon -> Show package contents -> Contents -> Eclipse -> open `eclipse.ini`
- find `Xdock:icon` there and check if it points to correct file. In my case it's `-Xdock:icon=../Resources/Eclipse.icns` and this file exists under Resources folder

the other workaround for this issue may be unselecting 'Launch in background`, but I'd rather stay on other approaches to fix it. However, to unselect it:
- edit the launch configuration by clicking on the "wheel icon" in the launch bar 
[image]

- go to 'Common' tab and unslecet the option 'Launch in background'
[image]

ammaree, 2022-02-19:
Nope, Have not tried second option but instruction does not tie up...
[image]

sigmaaa, 2022-02-19:
@ammaree have you tried to replace your -Xdock line with -Xdock:icon=../Resources/espressif.icns?

ammaree, 2022-02-19:
Will try tmrw

ammaree, 2022-02-21:
Have tried, now get 2 errors at start of build...
[image]

ammaree, 2022-02-21:
@sigmaaa 

Is there any way to "install" the Espressif-IDE to enable the persistence of common options?

sigmaaa, 2022-02-21:
Hi @ammaree  

I'm not sure about this, I'll ask my colleagues about it. Regarding your problem, you can also try giving the full path to `espressif.icns` in the -Xdock line instead of a relative path, or even remove that line. I was able to reproduce this error when I manually entered the wrong path, the problem went away when I simply removed that line as well.

Also the option to `Launch in background` helped in this case.

ammaree, 2022-02-21:
OK, the build error are gone, but cannot save/persist any options...