espressif/idf-eclipse-plugin #486: New directory "esp_idf_components" in project tree. Why? Is it mandatory?

Maksons, 2022-02-16:
From plugin version 2.4.0 in project tree I see one more folder : "esp_idf_components" .
Would like to hear comments:
    - Why it is needed?
    -  Is it mandatory?
    - Could it contain links to components nor all those components?
    - Should I move my and updated by me components to it ?
    - are all of them needed and how to disable not neede ones?

Possible bug:
When modified esp component is in project "components" folder it do not get to "esp_idf_components", Good.
But, when "esp_idf_components" already contain component and newly modified component get placed in project "components" folder, "esp_idf_components" is not updated and still contains same component.

Example  "esp_http_server" modified one.
1) Good case
    - modified "esp_http_server" is in "components" folder
    - at first build new folder "esp_idf_components" and it is without "esp_http_server".
2) Bad case 
    - "components" folder without modified "esp_http_server"
    - at first build new folder "esp_idf_components" and it contain "esp_http_server"
    - modified "esp_http_server"  placed in "components" folder
    - after build folder "esp_idf_components" still contain "esp_http_server"

kolipakakondal, 2022-02-16:
> - Why it is needed?

`esp_idf_components folder` is introduced with v2.4.0 release to achieve source code navigation for the function definitions((Press F3 or Ctrl+Click on the functional call) and to fix unresolved header problems with the indexer. It will be created immediately after the build based on the `build/compile_commands.json` file list. secondly, it also helps the users who want to check source of esp-idf components directly from the IDE. The files created under esp_idf_componenets are _virtual folders_ for `/esp-idf/components` so if you modify something in the `esp-idf/components/esp_http_server/http_main.c` it will be immediately reflected in the eclipse editor after refresh.

> - Is it mandatory?

Yes, if you want function definition navigation. This could be disabled by adding the `-Dskip.idf.components=true` VM argument in the eclipse.ini file

> - Could it contain links to components nor all those components?

It contains links to all the files found in the `build/compile_command.json` file. This is generated by CMake build. Please check more on this [here](https://cmake.org/cmake/help/latest/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html).

> - Should I move my and updated by me components to it ?

It's autogenerated after every build based on the `build/compile_command.json` file.

> - are all of them needed and how to disable not neede ones?

Again, the source of truth for this is the `build/compile_command.json` file.

tsctrl, 2022-03-05:
it does not look nice and hurt my eyes, as it does not have a meaningful or reasonable to be added in the user project root and not belong to the user work either and it is not related. this should be in the build folder.

ammaree, 2022-03-05:
+1
It also complicates search results since items are often (always) found twice.
Please [re]move

Maksons, 2022-03-07:
+1 
to move it inside "build" folder

kolipakakondal, 2022-03-08:
Thanks for your suggestions and inputs. Let us check and revert back on this.

gtjoseph, 2022-07-18:
It also doesn't have the header files in it which kinda defeats the purpose. :)

closed 2022-08-23 by kolipakakondal