espressif/idf-eclipse-plugin #490: IDF 4.4 new localized components issue and missing declaration in sdkconfig

ralphde, 2022-02-18:
Due to debug issue with missing uxTopUsedPriority I looked into it and found FreeRTOS-openocd.c where this global is defined if CONFIG_FREERTOS_DEBUG_OCDAWARE is defined. I understand that this should be defined globally and hence missing in sdkconfig or, else, where should one define CONFIG_FREERTOS_DEBUG_OCDAWARE ?

However, I found that  the file FreeRTOS-openocd.c is in <project>esp_idf_components/freertos. I changed the the source to this:
`
//#ifdef CONFIG_FREERTOS_DEBUG_OCDAWARE
const int USED DRAM_ATTR uxTopUsedPriority = configMAX_PRIORITIES - 1;
//#endif
`
However, no new compile was initiated. so question: what are the esp_idf_components in the project space good for? I also recognized that all the header files would still remain in the IDF install path.

I appreciate that the components are now part of the project, allowing replacement of changes in the components without ruining the whole installation, but that goal does not seem to be there yet in 4.4 or do I oversee something, any setting etc.?

Thanks

ralphde, 2022-02-18:
No matter where and how I define uxTopUsedPriority = configMAX_PRIORITIES - 1 it seems to be discarded from the build. OpenOCD keeps complaining about missing it. Can somebody explain this?

kolipakakondal, 2022-02-22:
Hi @ralphde 

Please find the details [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#why-esp_idf_components-folder-is-created-in-the-project) on why we have created esp_idf_components as part of the project.