espressif/idf-eclipse-plugin #492: Serial Terminal is unavailable in the new Espressif IDE

tsctrl, 2022-02-20:
i was unable to see the Serial Terminal in the new IDE for v4.4 as below:
this is required to see the logs that was run on different Baud rate.

Eclipse Version: 2021-03 (4.19.0)
[image]

New terminal Espressif IDE Version: 2.4.0
Build id: 20220128-0745:
[image]

please help.
thanks!

sigmaaa, 2022-02-20:
Hi @tsctrl,

this serial terminal is part of the eclipse CDT. You can install it from Espressif IDE. To do it:
* In Espressif IDE click on Help -> Install New Software -> Work with: https://download.eclipse.org/tools/cdt/releases/10.4 -> Select Terminal Features -> Install -> Restart IDE
[image]

let me know if this helps. Thanks

tsctrl, 2022-02-20:
ok, thanks for the prompt reply. close as resolved

closed 2022-02-20 by tsctrl