espressif/idf-eclipse-plugin #494: File ".project" is modified uncontrolled

franz-ms-muc, 2022-02-23:
**Describe the bug**  
 

working on Windows: 

```
 </natures>
<linkedResources>
 <link>
  <name>esp_idf_components/app_trace/app_trace.c</name>
  <type>1</type>
  <location>/C:/Users/Franz/esp-idf-master/esp-idf/components/app_trace/app_trace.c</location>
 </link>
```

working on LInux

```
<location>home/franz/C::/Users/Franz/esp-idf-master/esp-idf/components/app_trace/app_trace.c</location>
```

Working on Windows: 

```
<location>C:/home/franz/C::/Users/Franz/esp-idf-master/esp-idf/components/app_trace/app_trace.c</location>
```

**To Reproduce**  
 

open the Project in Windows, open in Linux, Open in Windows.

kolipakakondal, 2022-02-24:
Hi @franz-ms-muc 

Thanks for raising this issue. I think we need to find a better way to manage the` esp_idf_components` file references. The files under this folder will be refreshed after every build based on the compile_commands.json file. As you are building in different platforms so file system references are getting changed.

I understand this is a real concern let us think and work on this.

Thanks

franz-ms-muc, 2022-02-25:
How about setting it to the real Path ?

Not adding

Like C:

Home/c:

C:/home/c:

Why would one let this grow endless ?

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

diplfranzhoepfinger, 2022-06-29:
> Hi @franz-ms-muc
>
> Thanks for raising this issue. I think we need to find a better way to manage the `esp_idf_components` file references. The files under this folder will be refreshed after every [...]

no. 

if it would be "**changed**" all would be OK. 

you just append nonsense.

diplfranzhoepfinger, 2022-07-27:
> Hi @franz-ms-muc
>
> Thanks for raising this issue. I think we need to find a better way to manage the` esp_idf_components` file references. The files under this folder will be refreshed after every [...]

any news ???? 

please work on it so the Path is no more "extended" but deleted and re-generated.