espressif/idf-eclipse-plugin #495: Errors occurred during the build.

franz-ms-muc, 2022-02-23:
**Describe the bug**
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'Controller'.
Building Controller
/home/franz/git/ms/ISOAgLib_Versuche/Controller/build/Controller.bin
Building Controller
/home/franz/git/ms/ISOAgLib_Versuche/Controller/build/Controller.bin

flashing not possible. 
no information about the Error given. 

Output of the Build Console: 

```
Building in: /home/franz/git/ms/ISOAgLib_Versuche/Controller/build
cmake --build . -- -v
[1/3] cd /home/franz/git/ms/ISOAgLib_Versuche/Controller/build/bootloader && /home/franz/.espressif/tools/cmake/3.16.4/bin/cmake --build .
ninja: no work to do.
Build complete (0 errors, 0 warnings): /home/franz/git/ms/ISOAgLib_Versuche/Controller/build
/home/franz/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/franz/esp-idf-v4.3.1/tools/idf_size.py /home/franz/git/ms/ISOAgLib_Versuche/Controller/build/hello-world.map
Total sizes:
 DRAM .data size:   10812 bytes
 DRAM .bss  size:   26496 bytes
Used static DRAM:   37308 bytes ( 143428 available, 20.6% used)
Used static IRAM:   49918 bytes (  81154 available, 38.1% used)
      Flash code:  407875 bytes
    Flash rodata:  194196 bytes
Total image size:~ 662801 bytes (.bin may be padded larger)
/home/franz/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/franz/esp-idf-v4.3.1/components/partition_table/gen_esp32part.py /home/franz/git/ms/ISOAgLib_Versuche/Controller/build/partition_table/partition-table.bin
```

franz-ms-muc, 2022-02-24:
i try going back the Versions, which Version the Error occurs !

franz-ms-muc, 2022-02-24:
ESP-IDF Eclipse Plugin v2.4.1 bad
ESP-IDF Eclipse Plugin v2.4.0 bad
ESP-IDF Eclipse Plugin v2.3.0 good. running perfect !

franz-ms-muc, 2022-02-25:
same under Windows. 2.4.1 unable to flash a project and this annoying pop-up:

Maksons, 2022-03-08:
I think you have the same problem I reported in :
issue #485  and there is some solution.
Project name differ from file generated:
at picture :  Application.bin
il yours logs:  Controller.bin

franz-ms-muc, 2022-03-09:
you are totally right. 

there must be a match: 

1. CMakeLists.txt what is in "project"
2. Foldername
3. the Tag "name" in ".project" .. 

done !

closed 2022-03-09 by franz-ms-muc