espressif/idf-eclipse-plugin #496: Plugin install doesn't install the websocket-client package needed by the terminal

gtjoseph, 2022-02-25:
When you try to open a terminal the first time after a clean install of esp-idf and ide-eclipse-plugin, you get...
```
Traceback (most recent call last):
  File "/usr/src/mcu/esp32/esp-idf-v4.4/tools/idf_monitor_base/web_socket_client.py", line 57, in _connect
    self.ws = websocket.create_connection(self.url)
NameError: name 'websocket' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/mcu/esp32/esp-idf-v4.4/tools/idf_monitor.py", line 351, in <module>
    main()
  File "/usr/src/mcu/esp32/esp-idf-v4.4/tools/idf_monitor.py", line 299, in main
    ws = WebSocketClient(args.ws) if args.ws else None
  File "/usr/src/mcu/esp32/esp-idf-v4.4/tools/idf_monitor_base/web_socket_client.py", line 47, in __init__
    self._connect()
  File "/usr/src/mcu/esp32/esp-idf-v4.4/tools/idf_monitor_base/web_socket_client.py", line 60, in _connect
    raise RuntimeError('Please install the websocket_client package for IDE integration!')
RuntimeError: Please install the websocket_client package for IDE integration!
```
This should _really_ be taken care of by the installer or at least there should be clearer instructions without the backtrace.  For instance, you need to run `pip install websocket-client` from a shell that already has had the esp-idf export.sh script sourced.

kolipakakondal, 2022-03-17:
Hi @gtjoseph Thanks for reporting. That makes sense. We could install `websocket-client` also as part of the install tools process.

MichaelVLV, 2022-08-08:
Got this issue when switch sdkconfig `ESP System Settings->Panic handler behaviour`
from `Print registers and reboot` to` GDBstub on panic` and try run serial monitor in IDE

kolipakakondal, 2022-08-10:
Hi @MichaelVLV could you run `/idf/env/python/path -m pip install websocket-client` and see if that helps

MichaelVLV, 2022-08-11:
Hi @kolipakakondal , unfortunately doesn't helped.  I'm using Windows 10 and here is command that I've tried directly in ESP IDE:

```
C:\projects\ESP32\TOOLS\esp_idf_v44>python -m pip install websocket-client
Requirement already satisfied: websocket-client in c:\projects\esp32\tools\esptools\espressif\python_env\idf4.4_py3.10_env\lib\site-packages (1.3.3)

WARNING: You are using pip version 22.0.4; however, version 22.2.2 is available.
You should consider upgrading via the 'C:\projects\ESP32\TOOLS\ESPTOOLS\Espressif\python_env\idf4.4_py3.10_env\Scripts\python.exe -m pip install --upgrade pip' command.
```
I'm not sure about upgrading pip, wouldn't this break all modules?

kolipakakondal, 2022-08-11:
> Requirement already satisfied: websocket-client

This says websocket-client already installed.

MichaelVLV, 2022-08-11:
Rebooting PC seems helped to update this python lag.

But I've tested firmware again with selection **GDBstub on panic** at `ESP System Settings->Panic handler behaviour`.
Assertion happens in code and here is what I've got in log now.
```
в†ђ[1;31mWebSocket connection error: module 'websocket' has no attribute 'create_connection'в†ђ[0m
в†ђ[1;31mWebSocket connection error: module 'websocket' has no attribute 'create_connection'в†ђ[0m
в†ђ[1;31mWebSocket connection error: module 'websocket' has no attribute 'create_connection'в†ђ[0m

Traceback (most recent call last):
  File "C:\projects\ESP32\TOOLS\esp_idf_v44\tools\idf_monitor.py", line 351, in <module>
    main()
  File "C:\projects\ESP32\TOOLS\esp_idf_v44\tools\idf_monitor.py", line 299, in main
    ws = WebSocketClient(args.ws) if args.ws else None
  File "C:\projects\ESP32\TOOLS\esp_idf_v44\tools\idf_monitor_base\web_socket_client.py", line 47, in __init__
    self._connect()
  File "C:\projects\ESP32\TOOLS\esp_idf_v44\tools\idf_monitor_base\web_socket_client.py", line 65, in _connect
    raise RuntimeError('Cannot connect to WebSocket server')
RuntimeError: Cannot connect to WebSocket server
```

Not sure if this related to this issue or should be opened new one?
My IDF is v4.4:
_USR@DESKTOP-N85FKI5 MINGW64 /c/projects/ESP32/TOOLS/esp_idf_v44 ((v4.4))
$ git rev-parse v4.4
136475a9d9a0e2c6601ea94294e161a3ddb05810_

kolipakakondal, 2022-08-12:
Hello, Could you share the eclipse log to troubleshoot further on this?
https://github.com/espressif/idf-eclipse-plugin#error-log

MichaelVLV, 2022-08-12:
Here is my log:
[esp32_error_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/9316329/esp32_error_log.txt)

undergnd, 2022-11-16:
Hi, 

It looks like the issue still exist - I've just installed IDE, newest plugins and tried the above proposals of solution. When I use "idf.py -p COM2 monitor" in Windows 10 console it works as expected. In IDE i see the below:

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v4.4.3\tools\idf_monitor_base\web_socket_client.py", line 57, in _connect
    self.ws = websocket.create_connection(self.url)
NameError: name 'websocket' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v4.4.3\tools\idf_monitor.py", line 361, in <module>
    main()
  File "C:\Espressif\frameworks\esp-idf-v4.4.3\tools\idf_monitor.py", line 307, in main
    ws = WebSocketClient(args.ws) if args.ws else None
  File "C:\Espressif\frameworks\esp-idf-v4.4.3\tools\idf_monitor_base\web_socket_client.py", line 47, in __init__
    self._connect()
  File "C:\Espressif\frameworks\esp-idf-v4.4.3\tools\idf_monitor_base\web_socket_client.py", line 60, in _connect
    raise RuntimeError('Please install the websocket_client package for IDE integration!')
RuntimeError: Please install the websocket_client package for IDE integration!

Evanshion, 2022-11-18:
Same here.
I installed the websocket_client but I continue to receive the error.

sigmaaa, 2022-11-23:
Hi @undergnd, @Evanshion,

sorry for the late reply. We tried to reproduce this problem but were unable to. Also, we added installing WebSocket packages directly into the Install Tools process, so my guess is you having this issue because something is going wrong during tools installation. Can you provide the install tools log, so we could check, please? To install tools from IDE:
`Espressif -> ESP-IDF Tools Manager -> Install Tools`

> It looks like the issue still exist - I've just installed IDE, newest plugins and tried the above proposals of solution. When I use "idf.py -p COM2 monitor" in Windows 10 console, it works as expected. In IDE I see the below:

It can be because a different python environment (check <user_name>/.espressif/python_env path) is set for these processes. 
You can also check  environment variables inside IDE:
`Preferences -> C/C++ -> Build -> Environment -> IDF_PYTHON_ENV_PATH` 
Then check if python from IDF_PYTHON_ENV_PATH contains websocket site-packeges. For example, in the folder `.espressif\python_env\idf5.1_py3.8_env\Lib\site-packages` I have 2 folders related to WebSocket `websocket` and  `websocket_client-1.4.1.dist-info`

Evanshion, 2022-11-23:
Hi! I did as you suggested:  Espressif -> ESP-IDF Tools Manager -> Install
Tools
 And it worked!
Thank you very much for support!
BR

kolipakakondal, 2024-04-15:
Closing this issue we have already addressed this

closed 2024-04-15 by kolipakakondal