espressif/idf-eclipse-plugin #499: IDF 4.4 projects interfere symbol resolving of each other

ralphde, 2022-03-04:
**Describe the bug**
I migrated a project from the previous IDF and Plugin version into 4.4

`Operating System: windows 10
Java Runtime Version: 16.0.2+7-67
Eclipse Version: 4.21.0.v20210906-0500
Eclipse CDT Version: 10.4.0.202106130052
IDF Eclipse Plugin Version: 2.4.1.202202151702
ESP-IDF v4.4
Python set for IDF_PYTHON_ENV: Python 3.10.2
Python declared in PATH variable: Python 3.10.2
`

then I added the sdspi sample project into the workspace. all was fine and error fee compile. But then suddenly symbols started not to be resolvable in the UNTOUCHED sdspi sample project. 

Trying to nail down the issue I learned that I have to run sdkconfig (open AND save to invoce rebuild) for every project in the workspace every time I switch to it.

Notable as well that deleting the build directory does not have the same effect. I do have to open and SAVE skdconfig even after a complete rebuild from deleting the build directory.

notable as well that I had the impression that enabling the legacy header file inclusion made this a bit better and at least reproducible issue
   

 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

kolipakakondal, 2022-04-04:
[image]

Hello, Sorry for the delay in responding. I was trying to check with the plugin version 2.4.2 and IDF 4.4, but I don't see any symbols unresolved issues. Please find the screenshot for the sdspi sampe project .

From v2.4.0 onwards, the idf eclipse plugin has added esp-idf-components to the project because of that indexing might take some time to index and resolve the headers.

If you still have the same problem, do let us know so that we could investigate further.

ralphde, 2022-04-04:
That is not the right test. You need to have more than one project in the
workspace The resolve issue appears in the other project once you compile
the active one. The issue is that to fix the resolve you need to build from
scratch to get rid of it or better the build process is everytime over all
source files.
That means every time you switch between projects you vaste time for the
build. Very uncomfortable and time consuming if you change only one source
file in each project
Best Regards

Mit freundlicher Empfehlung

Ralph Deffke