espressif/idf-eclipse-plugin #501: Failed to build blink project with specific version in CMake

tsctrl, 2022-03-05:
**Describe the bug**
Unable to build blink project using Espressif IDE 2.4, i am able to build the same project using old ESP-IDF Eclipse 2021-03

**To Reproduce**
Steps to reproduce the behavior:
1. Import project as attached here 
[blink.zip](https://github.com/espressif/idf-eclipse-plugin/files/8190963/blink.zip)
2. Build
3. See error

**Expected behavior**
Should be able to build

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20211111 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
  
**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
ESP-IDF Eclipse Plugin
Version: 2.4.0.202201280731

**Eclipse Error log:** 
Window > Show View > Error Log
java.nio.file.NoSuchFileException: C:\Users\eclipse-workspace\blink\build\blink.bin
at java.base/sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:85)
at com.espressif.idf.launch.serial.internal.SerialFlashLaunchConfigDelegate.buildForLaunch(SerialFlashLaunchConfigDelegate.java:249)

Thank you!

tsctrl, 2022-03-06:
Update to Version: 2.4.1.202202101602 does not fix this issue

Maksons, 2022-03-08:
Possible same problem as reported in #485 , there is possible solution.
It is in case your project name in tree differ from hardcoded "blink" in CMakeLists.txt

tsctrl, 2022-03-08:
probably not the same,  that issue refers to project name different with the compiled binary. but in my case. the compiled binary is generated with the correct version but flash failed as it looks for the bin as the project name

sigmaaa, 2022-03-11:
Hi @tsctrl,

thank you for reporting this issue. It should be fixed for [2.4.2](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.2)

tsctrl, 2022-03-12:
fixed in 2.4.2

closed 2022-03-12 by tsctrl